LumaTeam/Luma3DS #401: Black screen latest nightly 11.4 firmware

teuf504, 2017-04-15:
Hello

I updated Luma to the latest nightly in order to update my firmware to 11.4.

I get the purple menu from Luma whenever I boot my o3DS (as usual) but when I launch boot.3dsx I get stuck in a non-ending black screen.

Anyone might know how I can get me un-stuck?

Margen67, 2017-04-15:
What entrypoint are you using?

teuf504, 2017-04-15:
OoTHax to install menuhax since my o3DS was unwilling to use menuhax from the get go.

I don't use soundhax so it shouldn't be a problem with the new firmware based on what I read

guiscaranse, 2017-04-15:
There is no *hax support for 11.4 yet

Margen67, 2017-04-15:
https://3ds.guide/

teuf504, 2017-04-15:
@Margen67 A9LH is installed on my o3DS since I updated to Luma though. The last part of the guide is completed. The only difference is that I still use an emuNAND rather than the sysNAND because I prefer to do so

Margen67, 2017-04-15:
If you have A9LH why did you say

>OoTHax to install menuhax since my o3DS was unwilling to use menuhax from the get go a year or two ago.

>I get the purple menu from Luma whenever I boot my o3DS (as usual) but when I launch boot.3dsx I get stuck in a non-ending black screen.

If you have A9LH you should just use the HBL CIA.

>Then Luma isn't too, to the contrary of what's written in many places, since a9lh is also a *hax

what

teuf504, 2017-04-15:
> If you have A9LH why did you say
Because I use the boot.3dsx meant for menuhax that's in the Luma3DS archive. Hence the fact I say I'm using menuhax as an entry point

> If you have A9LH you should just use the HBL CIA.
I can't

> what
It's called arm9loaderhax so it's a *hax. If no *hax is supported on 11.4, then Luma shouldn't work at all, which isn't the case unless I'm mistaken

cheatfreak47, 2017-04-15:
@teuf504 
First of all Arm9LoaderHax is not a *hax. It's a special entrypoint for running Arm9 code at boot, not running userland Arm11 homebrew. (Which is what we call *hax in this scene)

Because you're over here spouting off only random bits of information about your 3DS, we're unable to understand what exactly your 3DS's setup is like and can't help.

You shouldn't be using MenuHax, nor should you be getting any kind of "Purple Menu" unless you are running with an extremely outdated setup- meaning you **don't** have arm9loaderhax installed.

If you have Arm9LoaderHax, you should be running Luma 3DS with a CFW SysNAND, which should be updated to at least 11.3 by this point. If you need to launch .3dsx homebrew or run the Homebrew Launcher, you would be doing that by running Yellows8's Homebrew Launcher Loader (which is installed as part of 3ds.guide)

3ds.guide sets you up with Luma 3DS booting as a Arm9LoaderHax Payload- and if you have Arm9LoaderHax, all you do to update Luma yourself is replace the old "arm9loaderhax.bin" on the SD card root with the newest one. Hamcha's Luma 3DS Updater does this on-console without the need for a computer or anything.

Lastly, from the sound of things, this is a USER issue, not a problem with the Luma3DS software.

gnmmarechal, 2017-04-15:
They probably think they have A9LH because the payload is named arm9loaderhax.bin . Oh well.

MelonGx, 2017-04-16:
All entrypoints use arm9loaderhax.bin since 6.3:
- MSET -> Luma3DS.dat (6.3.1) -(redirect)-> arm9loaderhax.bin
- Spider -> Luma3DS.dat (6.3.1) -(redirect)-> arm9loaderhax.bin
- Boot Manager for *hax -> Luma3DS.dat (6.3.1) -(redirect)-> arm9loaderhax.bin
- boot.3dsx (Luma) -> arm9loaderhax.bin
- A9LH -> arm9loaderhax.bin
- A9LH -> Boot Manager for A9LH -> arm9loaderhax.bin

gnmmarechal, 2017-04-16:
Yep, but this misconception is fairly common.

AuroraWright, 2017-04-17:
Fixed

closed 2017-04-17 by AuroraWright