LumaTeam/Luma3DS #404: [REQUEST] Update issues layout

Jolty95, 2017-04-15:
Just a reminder to update it :P

Margen67, 2017-04-15:
https://github.com/AuroraWright/Luma3DS/wiki/FAQ-and-troubleshooting#i-get-an-auto-shutdown-with-gbadsdsiware-games also needs to be updated

ihaveamac, 2017-04-27:
@Margen67 what needs updating in this section?

Margen67, 2017-04-27:
@ihaveamac I think you addressed and fixed everything.

ihaveamac, 2017-04-27:
I'm not sure what in it needs fixing. 3DS Guide's Troubleshooting page doesn't cover agbfirm.

Margen67, 2017-04-27:
https://github.com/AuroraWright/Luma3DS/wiki/FAQ-and-troubleshooting#i-get-an-auto-shutdown-with-gbadsdsiware-games only covers TWL_FIRM.
Not TWL Version Data, DS Internet, DS Download Play, or Nintendo DS Cart Whitelist which also help solve TWL problems.

ihaveamac, 2017-04-27:
a link could probably be added but it won't be entirely replaced with it

Jolty95, 2017-04-27:
Now it's updated :P

Thansk @ihaveamac  :)

closed 2017-04-27 by Jolty95