LumaTeam/Luma3DS #405: SMILE BASIC Crashes Loader

gudenau, 2017-04-16:
When I launch SMILE BASIC v3.3.1 on FW 11.4.0-37U Loader calls svcBreak.

An exception occurred

Processor: ARM11 (core 1)
Exception type: prefetch abort (svcBreak)
Current process: loader

R0 00000001 R1 FFF9180C
R2 FFF90F70 R3 FFF9CF1C
R4 D9004587 R5 0FFFFF78
R6 140058C8 R7 00000001
R8 0FFFFEE8 R9 14005470
R10 00000D54 R11 00000001
R12 00000000 SP 0FFFFE20
LR 18001BC8 PC 140034E8
CPSR A0000010 FPEXC 00000000
You can find a dump in the following file:
dump/arm11/crash_dump00000020.dmp

Press an button to shutdown

HugaShire, 2017-04-16:
That is because 11.4 breaks Homebrew (if you have not updated payloads). This is a homebrew error not Luma3DS

gudenau, 2017-04-16:
How is it a homebrew error? I am not attempting to use homebrew.

Margen67, 2017-04-16:
What version of Luma are you using?

gudenau, 2017-04-16:
7.0

AuroraWright, 2017-04-17:
Do you have any game patching feature enabled for the title? LayeredFS, or locale.txt, or code.ips...?

AuroraWright, 2017-04-17:
Can't reproduce, game works fine even with LayeredFS redirection and language emulation. You are probably running it on an old NAND/FIRM.

closed 2017-04-17 by AuroraWright