LumaTeam/Luma3DS #407: Booting Luma v7 through homebrew launcher

arubar0, 2017-04-16:
Using the files provided inside the "hax" folder to boot Luma through menuhax/homebrew launcher leads to a black screen (you cannot even boot the config screen).

Using the files provided on v6.6 instead solves the problem though.

andamirorulez, 2017-04-16:
I have the same problem
N3ds, sysnand 9.2, menuhax
I use the 3dsx luma to direct boot
Black screen with luma 7
Works perfect if back to 6.6

arubar0, 2017-04-16:
@andamirorulez Try using all the files you need from v6.6 and then replace only the arm9loaderhax.bin with the one provided on v7 and tell us if you're able to boot Luma v7

Margen67, 2017-04-16:
https://3ds.guide/

andamirorulez, 2017-04-16:
@arubar0 i gonna try that, copying only the arm9loaderhax.bin from v7
I think boot.3dsx from v7 is corrupt

@Margen67 can you tell me exactly where to read?

AuroraWright, 2017-04-16:
[out.zip](https://github.com/AuroraWright/Luma3DS/files/924296/out.zip)
Does this work?
And no, not corrupted, but updating ctrulib leads to random cakebrah breakages

Margen67, 2017-04-16:
@andamirorulez Click "Get Started"

andamirorulez, 2017-04-16:
@Margen67 i use menuhax, not arm9
So that guide its not for me
That why i asked you where tu read in the guide

@AuroraWright i gonna try that later and feedback thanks!

Margen67, 2017-04-16:
>i use menuhax, not arm9

Why?

MelonGx, 2017-04-16:
@Margen67 For professional usages.

BTW, Luma3DS.dat after 6.3.1 is always unable to re-direct to arm9loaderhax.bin.
If anyone need to use (1) Boot Managers for Menuhax or (2) MSET or (3) Spider, he/she should use 6.3.1's Luma3DS.dat instead of any later version's.

SkittleDash, 2017-04-16:
I'm having the exact same problem. This really needs to be fixed. I was looking forward to its new features. ;-;

Margen67, 2017-04-16:
@SkittleDash If you want to use the newest Luma without waiting for a fix for menuhax (which you shouldn't be using anyway) you can install A9LH https://3ds.guide/

AuroraWright, 2017-04-16:
Could anyone test the files I posted?

SkittleDash, 2017-04-16:
@AuroraWright Just tested. Still got a black screen.

arubar0, 2017-04-16:
@AuroraWright Not working for me either

SkittleDash, 2017-04-16:
@Margen67 I refuse to use A9LH because I just cannot be bothered to install it. I don't even care about boot time anyway. And I leave my sysNAND and emuNAND unlinked so my sysNAND will remain untouched.

maorninja, 2017-04-16:
@Margen67 No.

A9LH isn't the answer to their problem. That's a **workaround**. I don't care if it's a good workaround. Besides, if everyone here automatically made the jump to A9LH, then there won't be any testers for MenuHax, and Luma is trying to be "not limited to A9LH, AKA being available to multiple entrypoints".

MelonGx, 2017-04-16:
As what I said, Luma3DS.dat after 6.3.1 are all broken so everyone using CBM/Spider/MSET should use 6.3.1's Luma3DS.dat instead.
So how about trying old version's boot.3dsx?

SkittleDash, 2017-04-16:
@MelonGx I can confirm that this worked. Thanks! ^_^

MelonGx, 2017-04-16:
@SkittleDash Good to hear that.

I still wonder what @AuroraWright has updated on boot.3dsx since 6.6 & Luma3DS.dat since 6.3.1 which cause those files nonfunctional.

EightyVI, 2017-04-16:
@AuroraWright Not work for me. Black screen in emunand.

pbanj, 2017-04-16:
@SkittleDash it should be noted that just because you have your nands unlinked doesn't mean your sysnand is safe. I've fixed a handful of systems that were using menuhax and an Emunand but somehow their sysnand ended up bricked. They were not in sysnand at the time, all of them were playing a game.

Not trying to scare you, just letting you know you're not as safe as you think.

AuroraWright, 2017-04-17:
It is fixed in the latest nightlies

andamirorulez, 2017-04-17:
@AuroraWright where can i download de nightlies? Are you talking about the releases?
Do you update the 7z file of v7??

pbanj, 2017-04-17:
http://astronautlevel2.github.io/Luma3DS/

closed 2017-04-17 by AuroraWright