LumaTeam/Luma3DS #415: Blackscreen on Launch (7.0)

Nariod14, 2017-04-17:
The new 7.0 update has quite a huge bug, it blackscreens on launch for me. I have an Old3ds (European) and I've followed your guide before to get Luma3ds working in my system, so it's definitely this update that's the issue. Menuhax will launch the bottom screen has it's little lightshow ending on red as usual then... nothing. It just stops there. I went running around the internet thinking I had messed with the wrong files (spending my entire afternoon beginning the steps to the infamous 3ds guide website thinking I had to do it now to fix it.) But little did I know, the only thing I had to do was roll back to 6.6. At your earliest convenience can you please fix this, alot of the features on the list seem very useful.

Margen67, 2017-04-17:
https://3ds.guide/

AuroraWright, 2017-04-17:
7.0.1 fixes it

closed 2017-04-17 by AuroraWright