LumaTeam/Luma3DS #416: Bootloop to Blackscreen (7.0.1)

Nariod14, 2017-04-18:
Hello again, firstly I'd like to thank you for your rather fast update(and your diligent work on this project), sadly though I've encountered another issue I'd like you to be aware of. Version 7.0.1 behaves once again, rather strangely on launch: when launched it would boot up once again with the lightshows, then shutdown, reboot to finally freeze at yet another forever lasting blackscreen. I've updated both manually and using the LumaUpdater tool (really helpful btw) but alas, the same issue occurs. Seeing as you've fixed the previous blackscreen issue I may be the only one to report this but I'm fine with staying at 6.6 until you've the time to patch this. 

Specs:
Old 3ds (European)
Sysnand: 9.2.0-20E
Emunand: 11.4.0-37E (latest)
Entry Point: Menuhax

Reading through the other issues it seems some have an aversion to menuhax, preferring instead to use the other guide for arm9loaderhax (something like that) might I ask if you plan to migrate to supporting only that or whether menuhax support will continue?

Margen67, 2017-04-18:
https://3ds.guide/

pbanj, 2017-04-18:
From the sounds of it you have menuhax installed on your Emunand and it is also set to auto trigger, or you told it to auto boot sysnand. So it's booting, running luma, when it loads it is booting menuhax again and locking the system up. This isn't a luma issue but user error

Nariod14, 2017-04-18:
Okay so I went into the select config menu and as it turns out you are correct, auto boot to emunand was checked. I unchecked it and it worked, my mistake feel free to close this...

... also that was rather fast...

cheatfreak47, 2017-04-18:
@Nariod14 You can close it yourself, and consider moving your EmuNAND to SysNAND and switching to A9LH, it's a lot better.

Glad you got it sorted out.

Nariod14, 2017-04-18:
Will do thanks!

closed 2017-04-18 by AuroraWright