LumaTeam/Luma3DS #417: Outdated Wiki - Region/Language Emulation Path

evandixon, 2017-04-18:
The [Options & Usage](https://github.com/AuroraWright/Luma3DS/wiki/Options-and-usage) page of the wiki currently does not detail how to use the region/language emulation on the latest stable release. In the past, the path for the locale file was `/luma/locales/<titleID>.txt`, but now it's `/luma/titles/<titleID>/locale.txt`. (It took me a bit to find out what the new path was, since I don't see it documented anywhere outside of [the code](https://github.com/AuroraWright/Luma3DS/blob/108e8a0cd408221d496ac47501338abd5c5327a4/injector/source/patcher.c#L458).)

It looks like there's other things on that page that might not apply to the latest version, but I can offer no specifics beyond the locale path.

Margen67, 2017-04-18:
The issue template also needs to be updated.

AceTrainer2712, 2017-04-18:
Does the emulation work with nothing else applied after the appropriate path change? Or is there anything else that needs to be changed?

evandixon, 2017-04-18:
Game patching needs to be enabled (like before, but the option was renamed IIRC).

AceTrainer2712, 2017-04-18:
Right, so I take it the archive patch referred to is the locale file?

sora10pls, 2017-04-18:
@AceTrainer2712 No. "Enable game patching" applies to LayeredFS, Code Replacement, Locales, etc.

dogtopus, 2017-04-18:
Few other things need to be updated too (e.g. the new chainloader)

AceTrainer2712, 2017-04-18:
@sora10pls yeah, I figured, but Luma says that to make game patching work, you need to apply an "archive patch", and I'm not sure what that means.

Margen67, 2017-04-24:
@TuxSH @AuroraWright

Margen67, 2017-04-24:
#404

yichof, 2017-04-26:
Hey guys, I think thre's still a problem even with the appropriate path change applied. I've followed the procedures described above and also enabled the game patching in Luma, and so far I've tried three different games (all US regions) on my JPN machine (N 3DSLL), and all of them stuck at the 3DS title loading screen permanently.

Could someone give me hand here? Thanks a lot!

urherenow, 2017-04-26:
Are you sure you have the seeds for thise games?

urherenow, 2017-04-26:
*those

ihaveamac, 2017-04-27:
https://github.com/AuroraWright/Luma3DS/wiki/Options-and-usage

closed 2017-04-27 by ihaveamac