LumaTeam/Luma3DS #419: Region free download play won't work on luma v7.0 and v7.0.1

stevenrockss, 2017-04-18:
I have a 3dsxl and an n3ds both US and a European 2ds, all CFW with a9lh and the latest luma nightly with system firmware 11.2 (lol was afraid to update to 11.3) anyways as the title says I can't use the region free download play with mk7 what do I do? Also, how do I switch from the nightly build of luma to the stable release?

adivonglong, 2017-04-19:
You need an emunand to be brave. LOL
Just replace the arm9loaderhax.bin or use the updater to update luma.

Nucca187, 2017-04-19:
so theres absolutely no way I can get Luma if I updated to 11.4 37 u
grr this is aggravating swapdoodle is set but have no one to send me doodlebomb just want Homebrew back my precious my precious homebrew.. jk just got off work was hoping for some good news on HBL

deadair3210, 2017-04-19:
If you have a9lh, just take a backup of your NAND and update to 11.4, won't hurt anything to my knowledge and if it does you can just restore your backup.
You can just download the latest version of Luma from the releases on this website and replace the files when prompted, or you can use something like Luma Updater.

EmuNAND is not required to be brave, as anything that can hurt you on a9lh SysNAND enough to brick will do the same on EmuNAND

adivonglong, 2017-04-19:
Fair enough, but restoring SysNAND isn't that easy and it's more risky.

Margen67, 2017-04-19:
@adivonglong How is it "not that easy"? You literally just choose a different option in D9 to restore sysNAND. Stop spreading misinformation.

Nucca187, 2017-04-19:
what if I didn't have a9lh all I was using was soundhax and updated without thinking like a nite like this after work....

Margen67, 2017-04-19:
@Nucca187 If you updated to 11.4 on sysNAND without A9LH without any sort of NAND backup there is no possible way you can downgrade.

Nucca187, 2017-04-19:
just making sure.. had to ask because no one is sending me doodlebomb. ty for taking the time to reply back though [😊]

adivonglong, 2017-04-19:
@Margen67 Please relate to the context, you wouldn't have D9 then.

urherenow, 2017-04-19:
Wut? Wasn't D9 around BEFORE a9lh? Either way, there has always been a .3dsx version.

AuroraWright, 2017-04-27:
This feature makes it possible to use Download Play with foreign games, it doesn't allow it between games or consoles from different regions from what I could see

closed 2017-04-27 by AuroraWright