LumaTeam/Luma3DS #420: I can't boot on EMUnand (Old 3DS)

CorJoestar, 2017-04-18:
I followed step by step the guide in this page ( https://3ds.guide/ ) but in the end I can't boot the Old3DS on EMUnand... 
I've installed Luma, but if I boot the device while I'm holding L button (with SysNand autoboot enabled) or otherwise (with SysNand autoboot disabled), it goes Sysnand at the same.

Margen67, 2017-04-18:
The guide has you transfer your emuNAND to your sysNAND.

pbanj, 2017-04-19:
Did the system ever even have an Emunand? If there is no Emunand on the SD card luma will boot sysnand every time as there is nothing else to boot

Kaisogen, 2017-04-20:
If you're asking how to boot an EmuNand, just format it using D9 or something else. Then copy your files back. Run an emunand then.

closed 2017-05-23 by TuxSH