LumaTeam/Luma3DS #421: LayerFS make home menu crash

Jolty95, 2017-04-18:
I hate to bother but I have a svcBreak prefetch abort exception when pressing HOME.
I have a romfs folder just like: `SD:\luma\titles\TITLE ID\romfs\` (I'm using a decompiled romfs in a folder just trying to crash Luma3DS and.....it crash when I press HOME button!

Here is the dmp  :P  

(Extra data: that romfs folder is a full romfs image of a translation that has been extracted in romfs folder in case my English is not very understandable)

[crash_dump_00000001.zip](https://github.com/AuroraWright/Luma3DS/files/930765/crash_dump_00000001.zip)

sora10pls, 2017-04-18:
#410

Jolty95, 2017-04-18:
@sora10pls I'm not talking about any system title...

sora10pls, 2017-04-18:
my b I misread lmao

Jolty95, 2017-04-18:
@sora10pls np :P

zuzma, 2017-04-19:
Not too sure what you're asking, but I've been running the beyond the labyrnith translation off of 7.0.1 with no problems at all (dumped and patched from my original card). Home menu works just fine too. 

used this guide to do it: https://gbatemp.net/threads/tutorial-how-to-decrypt-extract-rebuild-3ds-cia-cxi-files.383055/
Specs: N3DS A9lh luma3ds 7.0.1 on 11.4 firmware

AuroraWright, 2017-04-27:
I don't encounter this issue (I just tried with a translation of a JP game), are you sure the issue isn't in the translated files?

Jolty95, 2017-04-27:
Glad to talk again with you @AuroraWright  :)

I've tried with 7.0.3 and no dice.
I'm gonna tell you what I've tried so far (always with my original cartridge)

Kingdom Heart Dream Drop Distance + HANS romfs.bin Spanish translation on luma folder (correctly rename) -> No translation and crash on pressing HOME
Game + HANS romfs.bin extracted in romfs folder on luma folder (correct path ofc) -> No translation and crash on pressing HOME

Also, I've downloaded a CIA with the translation inside just to test (don't mean piracy because I have my original game! just wanting to test that romfs.bin)
Using romfs.bin from that cia, again, no translation and crash after presing HOME
Using the same romfs but extracted in romfs folder, the same, no translation and crash after presing HOME

If I dissable exceptions, It just hangs at 3ds logo

Also, don't know if this is related, trying to use a modified romfs from a homebrew (I mean the romfs folder from original source code), is a insta-crash if I try to open it

Thanks again for all your work :)

AuroraWright, 2017-04-27:
I just tried the Italian translation for the game and it works fine here (and Home doesn't crash). So you either didn't extract the RomFS files properly or the files themselves have issues.

Jolty95, 2017-04-27:
Thanks again but I've tried again with another download of romfs and extracted again (my SD is fully OK tested with h2testw) and still crashing when pressing HOME with this game (and translation not showing neither). Also, I've updated to Luma's c7e7dd82 and now in crash it reads:

Current proccess: menu (0004003000009802)

Really, I don't want to bother or nothing to you (and your time with this issue)

AuroraWright, 2017-04-28:
The game is the European version, right? I tried that one and the Italian fan translation via LayeredFS, it worked fine and Home works fine too. This means the issue is with the patch you are trying.

Jolty95, 2017-04-28:
Aurora I really trust you, really, don't think I want to fool you or something...
Would be possible to share me your translation just to test? Again, only if possible etc
Thanks for all!

Jolty95, 2017-04-28:
FIXED!!! 
Fiu... take me 7 hours but found the problem!!
I did RE to translation patcher.exe and get the romfs folder before compresing in romfs.bin. Later, diff the original extracted folder and the translated romfs folder and found the 6 files that were modified to be translated. Just copied that 6 files to luma and works without crashing!!

@AuroraWright two questions more:
-I only need that 6 files or do I need to copy ALL romfs folder to luma?
-Remeber what I said about homebrew's romfs folder? I've tried only with modified files and still crashing

Thanks again for all and sorry for that but at least I can play this awesome game in my language :D

gydisme, 2017-04-29:
sorry to add two more question here
1. is the LayerFS support CFW translation of romid/redirect.plg and folder/data.bin
for example, the zelda zelda mask of majora's translation files are `plugin/00040000000D6E00/redirect.plg` and `zelda/data.bin`

2. is the LayerFS support cheat codes?

AuroraWright, 2017-05-05:
You only need the modified files, and LayeredFS doesn't work with homebrew apps (it relies on functions only official games have).

closed 2017-05-05 by AuroraWright

Jolty95, 2017-05-05:
Thanks you @AuroraWright for answering :)