LumaTeam/Luma3DS #422: Why does homebrew freeze when i try to luanch stuff i have the 11.4 payload

RUNETHERABBITYT, 2017-04-19:
Like I have all the files but my homebrew just crashes on a green bottom screen and a black top screen and after 30 secinds my N3ds xl on 11.4 just turns off? is there a way to fix this or alternatively downgrade back to 11.3 by like useing the sysnand or whatever the hell its named

MelonGx, 2017-04-19:
Why you think that it's a Luma issue?

RUNETHERABBITYT, 2017-04-19:
Cus if it dosent crash it says luma3ds payload error

Zelec123, 2017-04-19:
Are you using a9lh for your entry-point? if not you may want to see if your using Luma 7.0.1.
[image]

RUNETHERABBITYT, 2017-04-19:
I updated to luma 7.0.1 and it still happens i even downgraded from 11.4 to 11.3 due to my SysNand from installing al9h

RUNETHERABBITYT, 2017-04-19:
Or whatever its called

TuxSH, 2017-04-20:
Use the 11.2 *hax payloads even on 11.3/11.4.

closed 2017-04-20 by TuxSH