LumaTeam/Luma3DS #424: Feature-Request: Disable 3DS Update Notification popup

Linden10, 2017-04-20:
There is a request that I like to ask, a feature request that I haven't seen anyone
done as of yet with homebrew/hacked 3DSes to this day.

I don't know how complicated this request maybe due to the nature
of 3DS NAND so this maybe simple or not...but I like to make a request
for a simple feature.

An option in Luma3DS to disable 3DS Update notifications.

There are ways, temporarily, to disable the 3DS Update notification by
going into the recovery menu on boot and pressing "cancel", deleting
the pre-downloaded update, but it only works temporarily and will comeback
as it re-downloads the update in the background.

I ask for this request simply because of how it can pop-up without
notice, almost leading to accidently updating by pressing a without realizing
it's popup. I almost anciently updated once in the past due to this notification
and I'm surprised no-one has enabled a way to disable it.

That's why I'm requesting this, I hope this isn't impossible
to do if at all. Thanks, I appreciate everything y'all have done
for making Luma3DS.

Margen67, 2017-04-21:
What entrypoint are you using?

Linden10, 2017-04-21:
By entrypoint you mean what I'm using t o load Luma CFW or
do you mean an entrypoint that can patch the notification?

If it's the former, I'm using arm9loaderhax.
As for latter...I don't have any nor jknow one. which is why I made this feature request
for a probably future implantation into Luma. I don't mind how long it may take for
this to be possibly, even if it takes months or longer.

Just felt like bringing attention to this since no-one has tackle it before.i

urherenow, 2017-04-21:
Wut...

Just block the update server on your router. Or use DNS settings that others have provided that does the same thing.

sora10pls, 2017-04-21:
Or... just, you know... update the 3DS?

MelonGx, 2017-04-22:
For EmuNAND users:
Why don't you
(1) Backup SysNAND
(2) Dump EmuNAND image
(3) Inject EmuNAND image into SysNAND
(4) Enter Recovery Mode to auto-delete the update nag
(5) Dump SysNAND then restore it into EmuNAND
(6) Restore SysNAND backup
?

For SysNAND users:
Why don't you upgrade to 11.4?

Linden10, 2017-04-22:
Hmm...it seems y'all believe this is about preventing updates to the newest version that
is 11.4. It isn't...and I appreciate the methods y'all have shared but I already know about them.

The DNS one can work but I recall it disables access to the eshop (Even on the newest version) and I read that updates can still get downloaded in the background through spotpass. If I wanted to disable that, I could but I use spotpass with a variety of things on my 3DS so I can't.

As for updating to the newest version, I was planning on to already but this isn't about that...
It isn't about me, it's about a means of disabling sysupdate notification or a method for Luma
to block updates if it goes that way.

Still thanks for helping and may this feature request, in some form or another, becomes possible
to implant in Luma in future versions. Yup!

Delphox, 2017-04-23:
A patch that intended to do this has been tried before but didn't work (see #242), so it was removed and the issue closed. We don't know enough about NIM to make it work yet, I guess.

But yeah, I agree with you, this would be a nice thing to have for various reasons if it was feasible. For the time being, what you can do is deleting the update (boot console in recovery mode, that is, boot holding L+R+DPad UP+A, then immediately cancel the menu), then avoid putting the console in sleep mode while connected to the internet. That'll stop the nag.

Nyquis, 2017-04-24:
+1
It would be nice to have a choice to decide if I want to update and disable updates. I think people telling to update to 11.4 are missing the point. If I stream with NTR then I would not want to update to 11.4

Another point is that ctr-httpwn exists. It would be nice to have an option in the select screen that makes these changes permanent and reverses them when disabled.

Margen67, 2017-04-24:
Good thing NTR is closed source so that can't be fixed

Kaisogen, 2017-04-24:
NTR is open source now? I thought Cell9 threw a fit and deleted everything. What did I miss?

Delphox, 2017-04-24:
It isn't, it was sarcasm from him (see meme face).

Kaisogen, 2017-04-24:
Hopefully there will be an alternative to it in the future. But for now, it is very unlikely. I just wanted screen recording and plugins anyways.

AuroraWright, 2017-04-27:
It was attempted a long time ago and there were weird counter-indications, so it wasn't pursued further

closed 2017-04-27 by AuroraWright