LumaTeam/Luma3DS #425: Region free patch failed to apply on luma v7.x

p1617762, 2017-04-21:
New 3ds US, 9.2 sysnand, 9.5 emunand.
Region free patch works on luma v6.6, but mset is broken.
Region free patch failed to apply on luma v7.x, but mset worked.
I even deleted the luma folder (so it can not be the path issue: https://github.com/AuroraWright/Luma3DS/issues/417)

jonirrings, 2017-04-21:
the locale path changed,
[code comment](https://github.com/AuroraWright/Luma3DS/blob/108e8a0cd408221d496ac47501338abd5c5327a4/injector/source/patcher.c#L458)

Margen67, 2017-04-21:
https://3ds.guide/

p1617762, 2017-04-21:
> the locale path changed,
I deleted the luma folder, how would locale path matter at all?

> https://3ds.guide/
I have a working CFW, just the region free patch failed to apply.

sora10pls, 2017-04-21:
>9.5 emunand

c r i n g e

p1617762, 2017-04-21:
> c r i n g e

I finished the 1 game I installed in 2015 just last month....
And 11.5 wasn't goodly supported during 7x nightly...

Margen67, 2017-04-21:
You shouldn't be using menuhax, mset, or soundhax. They're unsafe. You should install A9LH.

p1617762, 2017-04-21:
Still, I would like to get the region free patch working first. @Margen67

MelonGx, 2017-04-22:
1. @AuroraWright Wiki update seems to be the most important issue because there are more and more noobs asking for the latest region patch's path.

2. @p1617762 Read the fucking commit instead of the fucking wiki.
https://github.com/AuroraWright/Luma3DS/commit/b5336c81cc82b6c5e8115249342beb5b065cdce9

MelonGx, 2017-04-22:
How the fucking can region free patch work if you ignored what #417 required?

p1617762, 2017-04-23:
I used this:
> G:\luma\titles\0004000000166A00\locale.txt

with 

> JPN JP

but failed, that's why I removed the whole folder.
(Wait, so I can't pre-enable the region free patch ahead of time?) @Margen67

closed 2017-04-24 by p1617762