LumaTeam/Luma3DS #426: NSMB2 LayeredFS not loading mods

Gimzie, 2017-04-21:
The game New Super Mario Bros 2. does not load the modified romfs, and it's probably due to a different structure.

Multiple people have also reported this issue, and it would be way easier to load mods if the bug was fixed.

Thanks in advance!

AuroraWright, 2017-04-23:
Latest nightly fixes it

Gimzie, 2017-04-23:
Tested on latest hourly, seems to work! Thanks a bunch for your hard work! 'w'

closed 2017-04-23 by AuroraWright

maorninja, 2017-04-24:
@Gimzie has a Github? Did you just make it?

But anyways, @AuroraWright, why did those issues even occur to start with? (If I offended you and your work with this post, I'm sorry, K? I just don't know how to phrase stuff nicely....)

AuroraWright, 2017-04-27:
Because layeredFS works by patching the game's code and while I can test it with a limited amount of games there is a possibility the patch doesn't play nice with the code of *all games*.

shinra358, 2019-03-05:
not working luma 9.1