LumaTeam/Luma3DS #428: LayeredFS causes active theme to be disabled when exiting to home menu

jupitteer, 2017-04-23:
I've tried with multiple games and mods, and every time, exiting to the home menu causes the theme to be reset to default as if the theme extdata wasn't available. The themes can be re-enabled.

Margen67, 2017-04-23:
What entrypoint are you using?

jupitteer, 2017-05-10:
arm9loaderhax. It is the latest version of luma and arm9loaderhax.

AuroraWright, 2017-05-10:
https://github.com/AuroraWright/Luma3DS/issues/441

AuroraWright, 2017-05-10:
Quoting myself: I tried with an official theme (not via CHMM or other homebrew apps, but I don't think it should change anything and **if it does, it's not Luma's problem**), launched Pokemon Moon (both 1.0 and 1.1) with this romhack: https://projectpokemon.org/forums/forums/topic/40238-pok%C3%A9mon-nova-sun-umbra-moon-fully-featured-challenging-sunmoon-romhacks/ (which worked as expected) and home had no crashes or anything. **The theme also kept working as expected."**

jupitteer, 2017-05-10:
Trying with an official theme, it is not set back to no theme. Something with custom themes.

closed 2017-05-10 by jupitteer

Kaisogen, 2017-05-12:
I'm using a custom theme, with game patching for a Japanese Game (FE:Echoes Translation). I use a custom BoTW theme and this does not happen to me. It must be something with your specific theme. Can you post a link to it?