LumaTeam/Luma3DS #433: Adding ext: to layeredfs will also cause extdata to be redirected.

mariohackandglitch, 2017-04-27:
Many games mount the extdata archive as extdata: Allowing ext: to be redirected (only checking the e) will also cause the extdata files to be redirected. This may cause problems in the case where some extdata and romfs files have the same name (extdata:/data.dat and rom:/data.dat for example). 
I'm researching the archive type, and it looks like *(archiveptr + 0x18) always is a pointer to the stack if the archive is a romfs archive. I'll need some help to fully understand how the archive type works,  it's only a theory for now.

AuroraWright, 2017-04-27:
The problem is that we are space constrained in the payload. The code and the folder path (currently lf:/luma/titles/0000000000000000/romfs where the titleid is inserted) currently are at 275 out of 284 max.

AuroraWright, 2017-04-27:
I had an idea, I'll try implementing it

AuroraWright, 2017-04-27:
https://github.com/AuroraWright/Luma3DS/commit/db33c315f2fc65938ed4ea53a630355076b64e27

AuroraWright, 2017-04-29:
I had to revert in https://github.com/AuroraWright/Luma3DS/commit/1887c6fae3539200854b8c615932a1916d8fb591 because it breaks animal crossing and possibly other games, so the issue is still open. Did you discover an easy way to detect RomFSes from fsTryOpenFile @mariohackandglitch?

mariohackandglitch, 2017-04-30:
That's the only pattern that I've seen all romfs archives follow (for now). You could also hook to the function that mounts romfs (as it's different than RegisterArchive but I guess that function calls it), but it's asm pattern is unknown. It's also unknown if that function is used by all games.

TuxSH, 2017-04-30:
Could you provide the disasm of the function as well as the dump of the object, please?

mariohackandglitch, 2017-04-30:
I only know that function exists. Never had the opportunity to search it nor I have much asm knowledge. [Here are some archive dumps for some games.](https://mega.nz/#!KFc1gYBK!A6ZUPtfjkFH4Pz091vdWUKmWq-1iVzMeNU8R3BTe1Ms)

AuroraWright, 2017-05-05:
I think the bug has finally been fixed (and any other conflicts that may arise) in https://github.com/AuroraWright/Luma3DS/commit/e231275cbe304b546d7793283a6738f3fe6ed1ca

closed 2017-05-05 by AuroraWright

mariohackandglitch, 2017-05-07:
How was it fixed? Also, from what I can see you should also add pat1:/ as MK7 uses that for the update romfs