LumaTeam/Luma3DS #434: [Suggestion] Default resolution behavior for TWL_FIRM (NDS) and AGB_FIRM (GBA).

stargoonman, 2017-04-28:
Right now if you run an NDS game, you have to either hold START or SELECT to set it to its native resolution.

The screen may be smaller, but because it looks way better than horribly filtered screen when it's stretched to the 3DS resolution.

It would be awesome if we could set it as a behavior so we don't have to hold START or SELECT when we run NDS or GBA games.

Thanks!

Zetchzie, 2017-04-29:
It would be awesome to disable the filter so it doesn't look like crap to begin with in bigscreen mode

cheatfreak47, 2017-04-29:
@Zetchzie Get better GBA CIAs.

piratesephiroth, 2017-05-22:
-1

deadair3210, 2017-05-22:
Would it be possible to fix the big resolution to not be so bad looking? I mean the DS was no Picasso in the first place but the stretched res could use some work

Margen67, 2017-05-22:
@deadair3210 The filtering isn't the problem. The 3DS's screens don't have a high enough resolution to scale DS games by an integer (e.g. 2x) which is why it's needed in the first place.

cheatfreak47, 2017-05-23:
@piratesephiroth the proposal is to add it as an option to default to this behavior, it's just a convenience thing. Totally optional otherwise.

piratesephiroth, 2017-05-23:
it's already optional, just hold start or select

Margen67, 2017-05-23:
@piratesephiroth As in the option within Luma would be optional, and disabled by default.
Obviously the original feature is optional and disabled by default otherwise this issue wouldn't exist in the first place.

Kaisogen, 2017-05-24:
It really depends on the game.. I would like a feature like this, in this format:

A config file, which contains a list of the games you would like to activate it for.

so something like:

001293 [PvZ DS]
109349 [SM64DS]
149823 [Ace Attorney T&T]

and those would be at the original resolution automatically. 

Although would this require patching to TWL? Sounds like quite a bit of work.

TuxSH, 2017-06-05:
I think further research on TwlBg should be done to be able to have custom resolutions instead. Atm, you can press Start+Select, and I don't feel like implementing something with limited options.

Closing for now.

closed 2017-06-05 by TuxSH