LumaTeam/Luma3DS #435: Luma 7.0.4 Breaks this mod

Piefactor, 2017-04-29:
https://gbatemp.net/threads/release-animal-crossing-new-leaf-gcn-music-pack.468982/page-2#post-7277133

This mod replaces bcstm sound files of Animal Crossing New Leaf using LayeredFS.  For some reason it does not work and crashes the game on launch with 7.0.4, but does work with 7.0.3.

KunoichiZ, 2017-04-29:
I also posted this in the GBAtemp thread for Luma as well and can confirm this as well.

AuroraWright, 2017-04-29:
Is the game updated or not? If you enable exceptions is it loader the process that crashes or the game itself?

KunoichiZ, 2017-04-29:
The game is updated, for me at least, to v1.5. The game gets past the Nintendo 3DS loading then immediately crashes. I uploaded my crash dump in the Luma thread. @AuroraWright

AuroraWright, 2017-04-29:
https://github.com/AuroraWright/Luma3DS/commit/1887c6fae3539200854b8c615932a1916d8fb591 should fix it

KunoichiZ, 2017-04-29:
@AuroraWright Yep, that fixed it. Thanks.

closed 2017-04-29 by AuroraWright