LumaTeam/Luma3DS #437: [Feature Request] Custom String in System Settings can show Firmware Version?

DarkPulse, 2017-04-30:
Hello,

I got my N3DS moved off the EmuNAND yoke to Luma + A9LH last night, but it kind of makes me a bit edgy to see "Sys" as the string before the version number - even though I know it's safe to update, it's kind of hard to get used to the feeling of it being safe to update.

I found that there was a way to input a custom string, but when I did that, all that was displayed was the custom string - not the version number anymore,  which is a bit puzzling.

I did a search for that on the closed issues in this repository, and found one, issue #219, that says apparently it is supposed to show the version number; however, it also indicates a much shorter string limit/different format compared to now, so I'm presuming at some point, this got modified.

Maybe it'd be possible to add this back, even as some kind of user variable so that people who don't like it aren't saddled with it? i.e; I could make a string that's, say, "Custom %fw" and it'd display "Custom 11.4.0-37U" or whatever?

Petraplexity, 2017-04-30:
I believe if your custom string is >= 5 characters long, it will automatically show the version number after it. If it is < 5 characters long, it will show only your string. For example, a .txt file containing "1234 " will output "1234 11.4.0-37U". A .txt file containing "4.2.0-69 GRAND DAD" will output "4.2.0-69 GRAND DAD".
If that is still the case, this issue should be closed.

DarkPulse, 2017-04-30:
That wasn't mentioned in the FAQ/Wiki at all, but you are indeed right. Thanks for pointing it out! That said, you got your signs mixed up. :smile:  

If it's 5 characters or less, it will show the string plus the version number. 6 or more, it will show just the string, no version number.

But that's enough space to at least put "A9LH" in, so I guess this can be closed after all.

Maybe down the line it could be expanded, but hardly something that I'm going to fuss over when I can get effectively what I'd like for my sanity. :stuck_out_tongue:

closed 2017-04-30 by DarkPulse

Petraplexity, 2017-05-01:
Really? I swear I saw it somewhere on this Github.
Anyway, the limit has to do with how Nintendo programmed System Settings. It's only _supposed_ to display "Ver. " and then the version string. Hence, the <= 5 character limit for also showing the custom string along with the version string. I believe when Aurora was first implementing this, she discovered System Settings would crash if you tried to force a custom string > 5 characters to display simultaneously before the version string, and simply designed Luma's custom version string implementation around this.
**The More You Know!**