LumaTeam/Luma3DS #439: DS and gba games not working, already tried troubleshooting steps

wolflmg, 2017-04-30:
I'm having problems with DS and GBA games not working and just loading a black screen. I followed the trouble shooting files, but when i go to install the two cia files, it says the files already exists and wont let me install

Kaisogen, 2017-05-01:
Are you installing them through FBI? If the file exists already, FBI should simply overwrite it. Describe this in more detail.

Margen67, 2017-05-01:
Are you using an emuNAND?
What entrypoint are you using?

houodrigoly, 2017-05-07:
Margen67 I have the same issue, i'm using emunand, what you mean with entrypoint?

Bolt23, 2017-05-23:
Your TWL_FIRM must have died due to unnecessary doings by you or they must have been killed by you through peanut butte or there is some ghost 👻 who rigged your 3DS. Beware! Something is crawling on your 3DS! The Living 3DS is Dead! 

Taking Foolish like you right? Yes there might have been some serious problems when you installed A9LH or you updated to B9LH. Please specify your problems, and do tell that you have B9S or A9LH and no menuhax! Are you human or ******? ****** doesn't count anything, it is just a mere joke isn't it?

Margen67, 2017-05-23:
To run DS/GBA games from emuNAND they must also be installed to sysNAND.
If you're using menuhax/mset, follow [this guide](https://3ds.guide/).
If you're using A9LH: https://3ds.guide/updating-to-boot9strap

TuxSH, 2017-05-28:
^

closed 2017-05-28 by TuxSH