LumaTeam/Luma3DS #441: Pressing home button when game patching on Pokemon sumo freezes the game

slowbeatle, 2017-05-02:
Luma 7.0 to 7.0.4  11.4 a9lh n3dsxl. has this issue for me, only fix is disabling game patching or putting a stock color theme. Something related to themes seems to be the problem....or exdata

slowbeatle, 2017-05-02:
has anyone had this issue in any other games

slowbeatle, 2017-05-07:
the issue still persist in 7.0.5

AuroraWright, 2017-05-08:
I tried with an official theme (not via CMMM or other homebrew apps, but I don't think it should change anything and if it does, it's not Luma's problem), launched Pokemon Moon (both 1.0 and 1.1) with this romhack: https://projectpokemon.org/forums/forums/topic/40238-pok%C3%A9mon-nova-sun-umbra-moon-fully-featured-challenging-sunmoon-romhacks/ (which worked as expected) and home had no crashes or anything.

closed 2017-05-08 by AuroraWright

AuroraWright, 2017-05-13:
For the record, the issue has been confirmed to be caused by (certain?) custom themes. It does not occur with official themes from the theme store, thus it's not Luma's problem