LumaTeam/Luma3DS #443: [Suggestion] Service calls logging

natinusala, 2017-05-02:
Hello,

I don't know if it's possible or if it exists, but it would be nice to have a way to log all service calls (and their parameters ?) to a file in the SD card. It could be useful to see how system apps or close sourced homebrews work.

What do you think ?

TuxSH, 2017-05-02:
( ͡°( ͡° ͜ʖ( ͡° ͜ʖ ͡°)ʖ ͡°) ͡°)

TuxSH, 2017-05-02:
That kind of thing will be released soon enough, it just requires a lot of background work beforehand.

natinusala, 2017-05-02:
I see, thanks :)

dogtopus, 2017-05-04:
Meanwhile you could (ab)use citra to do this job if the victim app is simple enough

natinusala, 2017-05-04:
Well in my case that would be the System Settings app, or maybe some of the Nintendo SDK apps. I'm not sure if I can find those, or if Citra can run them, or even implemented the services I'm interested in...

natinusala, 2017-05-04:
I succeeded to convert the CIA to CXI and run it in Citra. How may I see the service calls, should I add a debug line to the source and recompile it ?

Pirater12, 2017-05-24:
Sorry for bumping this, but any updates?

natinusala, 2017-05-24:
In the meantime you can run the homebrew in Citra3DS, it effectively logs all known and unknown service calls in the console.

If you need additional informations, such as translate/static thread buffer values, you can easily recompile Citra to add such logging.

Pirater12, 2017-05-24:
Citra doesn't  have httpc and soc:P support

natinusala, 2017-05-24:
Well in this case I'm afraid that you'll have to wait for Luma to implement this feature =/

Kaisogen, 2017-05-24:
It certainly sounds interesting. But... How useful will this be? I can't see this being useful to most people in the future. If you know how to use these things, that's good.

But I don't see Aurora adding this instead of other features the average user is asking for. You'll probably want someone to make a fork to do this.

I don't see this being a part of the main program.

natinusala, 2017-05-24:
I'd say that it's as useful as disabling SVC checks or attaching the exception handler. It's to facilitate debugging homebrews, and certainly not for the average end user. 

The current state of Luma is fine because there are only two (three ?) options in the config menu. Eventually more options, for developers or not, will be added, and the config menu may become bloated. Maybe two branches of Luma will then be needed, one with dev options and one without.

Anyway I can understand if Aurora doesn't include this (or not in the priority list), it's not really a CFW feature, more like a developing quality of life feature.

Kaisogen, 2017-05-24:
Good point. There a few dev options on the config menu. 

I like the idea of two branches, a dev build and a non dev build.

But you could just as easily do something like SpriteTools debug menu, and have a line of code enable or disable the options to keep it from being bloated, and to help keep it into one build.

natinusala, 2017-05-24:
Having to recompile Luma just to enable dev features may be annoying, and is a loss of features compared to the current release. A menu category on the other hand is too overkill, but something like a key combo to make them appear can be a good compromise.

Pirater12, 2017-05-27:
[3dshax](https://github.com/yellows8/3dshax) seems to have IPC logging however it doesn't support sighax

natinusala, 2017-05-27:
There are tools to convert .bin payloads into .firm files, could that work or does it depend on A9LH to work ? However I'm not sure on how to use it, is it a CFW ? Should I use Luma chainloader to run it ?

Pirater12, 2017-05-27:
@natinusala 3dshax is a cfw for devs.  I'm pretty sure that some stuff needs to be changed in 3dshax before it  can work with sighax(I highly doubt that by just using firmtool to convert .bin->.firm is gonna make it work with sighax).

natinusala, 2017-05-27:
Well I should wait for yellows8 to update it then.

TuxSH, 2017-06-05:
(∩ ͡° ͜ʖ ͡°)⊃━☆ﾟ 

`monitor syncrequestinfo` with GDB on svc 0x32 entry. Doing this over the network may be a bit slow, I invite you to read the source code.

closed 2017-06-05 by TuxSH

natinusala, 2017-06-05:
You rock, thanks a lot !

natinusala, 2017-06-05:
For the record, this is how you use it in GDB once you're attached (thanks @TuxSH)

```
catch syscall 0x32 command 1
if($r0 > 0)
monitor syncrequestinfo
end
end
set pagination
off
continue
```