LumaTeam/Luma3DS #444: [LayerFS]  about CFW translations & cheats

gydisme, 2017-05-04:
sorry to ask two questions here

1. does the LayerFS support CFW translation of romid/redirect.plg and folder/data.bin
for example, the zelda zelda mask of majora's translation files are `plugin/00040000000D6E00/redirect.plg`  and `zelda/data.bin`

2. does the LayerFS support cheat codes?

dogtopus, 2017-05-04:
1. If that plg does exactly what LayeredFS does (i.e. redirect ROMFS files to files on SD card) then yes
2. No

gydisme, 2017-05-04:
I tried  the zelda : mask of majora's, I placed the plg file into `luma/titles/00040000000D6E00/romfs/redirect.plg` and also copied `sd: zelda/data.bin` , it's not work.

I launched the gateway mode to play the game, is it possible the reason why it's not working?

dogtopus, 2017-05-12:
You should put the patched game files used by your plugin, rather than the plugin itself, to romfs folder

gydisme, 2017-05-12:
I place the data.bin at luma/titles/id/romfs/, it still not work at all, it seems to me that is not a solution.

dogtopus, 2017-05-12:
Then the data is likely custom packed and is not supported by Luma

AuroraWright, 2017-05-13:
Launching gateway undoes Luma completely so you can't do that, also you don't need to copy the .plg but the romfs files (if data.bin is the romfs file, copy that one).

closed 2017-05-13 by AuroraWright

Kaisogen, 2017-05-14:
Is there any real advantage to gateway these days? I don't really see any reason to be using it, plus it directly implies piracy through its use.