LumaTeam/Luma3DS #449: Freezing issue when pressing home button (game patching related)

slowbeatle, 2017-05-12:
Playing pokemon sun with update 1.1 as cia, if i have wireless communication enabled and then press home button, the 3ds just straight freezes. N3DS XL Luma 7.0.5. Only happens when using game patching romfs.

Removing any installed theme and leaving wifi on solves the issue, turning of wifi solves the issue, deleting the 1.1 update solved the issue and removing any file from the a folder in my romfs solved the issue. For testing, i redownloaded both the game and update from the eshop, dumped pokemon sun and extracted it's romfs and placed the same files i modded, but as vainilla with the same folder structure, but i still get this problem related to themes/wifi on 1.1.

Gotta note that the files i'm replacing are unrelated to the 1.1 update, as this update didnt replace any of them (atleast for what i can see  when i extract the update)

https://drive.google.com/drive/folders/0B3PXxl29fzElWVBWREZsbk1zNUE
here's a link to the files i'm replacing

Margen67, 2017-05-12:
What entrypoint are you using?

slowbeatle, 2017-05-13:
A9LH

TuxSH, 2017-05-13:
Duplicate of https://github.com/AuroraWright/Luma3DS/issues/441

closed 2017-05-13 by TuxSH

AuroraWright, 2017-05-13:
Added a clarification https://github.com/AuroraWright/Luma3DS/issues/441#issuecomment-301245227