LumaTeam/Luma3DS #453: Crash when exiting System Settings

DarthVitrial, 2017-05-17:
**System model:** [e.g. 2DS, New 3DS, Old 3DS]
Old 3DS
**SysNAND version (+emu/redNAND version if applicable):** [e.g. 11.4.0-37U]
11.4.0-37U
**Entrypoint (How/what you're using to boot Luma3DS):** [e.g. A9LH]
A9LH
**Luma3DS version:** [e.g. 7.0.5 stable or if using nightly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/5fe83588eb4b8faafc7773b3739a3d3307160b04]
https://github.com/AuroraWright/Luma3DS/commit/e029458b09a2f2627a6e7aa6b425bebdd1aa3940

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (4 )

Splash: (Off )

PIN lock: ( Off)

--

Autoboot SysNAND: (yes )

Use SysNAND FIRM if booting with R: ( yes)

Enable loading external FIRMs and modules: ( no)

Use custom path: (no )

Enable game patching: (yes)

Show NAND or user string in System Settings: (yes )

Show GBA boot screen in patched AGB_FIRM: (no )

Patch SVC/service/archive/ARM9 access: (no )

Set developer UNITINFO: (no)

Enable exception handlers: (yes)

--

System crashed (prefetch abort, I think) 

when exiting the system settings. Not 100% sure it was a Luma issue, but that seemed most likely.
Crash dump: 
https://mega.nz/#!8slyHRrK!uBbjkp-_BYQYJs8GU8hmD6mQSBxsIthdq11YUWbq4DE

urherenow, 2017-05-17:
If you're not 100% sure, then you didn't even test. Why would you post a report on something that may have simply been a glitch? You need to be able to REPEAT a crash to say that 1) there is indeed a problemand 2) How to reproduce it.

I'm here to tell you to try and repeat it exactly, and if it happens again, turn off the "show nand or user string" setting and try again.

Kaisogen, 2017-05-17:
I agree with urherenow. Try reproducing the glitch, and see if it's consistent. Tell us any other potential info about your system we don't know. Or how you led up to it.

AuroraWright, 2017-05-17:
This happens with stock too, if you boot DS internet, quit and then exit system settings home menu crashes; but it's masked by the fact the console reboots. Luma "unmasks" it by showing the exception. It's actually the way I test the exception handler for arm11 :)

closed 2017-05-17 by AuroraWright

DarthVitrial, 2017-05-17:
Ah, yep, that's precisely what triggered it. Glad to know it's not a bug. :)

Kaisogen, 2017-05-17:
Yeah, I knew about HM rebooting (I think you can do a soft reboot by holding A with luma?), but didn't know about the crash after exiting DS Internet.

3DS Trivia of the day!