LumaTeam/Luma3DS #456: Windows Defender Blocks Download -

GriffinG1, 2017-05-20:
**System model:** [e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):** [e.g. 11.4.0-37U]

**Entrypoint (How/what you're using to boot Luma3DS):** [e.g. A9LH]

**Luma3DS version:** [e.g. 7.0.5 stable or if using nightly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/5fe83588eb4b8faafc7773b3739a3d3307160b04]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot SysNAND: ( )

Use SysNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch SVC/service/archive/ARM9 access: ( )

Set developer UNITINFO: ( )

Enable exception handlers: ( )

--

**Explanation of the issue:**

http://imgur.com/KskVy3n

**Steps to reproduce:**

1. Download Luma3DS.7z

2.

**Dump file:**

Rebelgaming12, 2017-05-20:
I have the same issue

jonirrings, 2017-05-20:
I just remove it from blocked list

AuroraWright, 2017-05-20:
It's obviously a false positive, Luma is open source so you can compile it yourself if you don't trust us.
Complain to microsoft.

closed 2017-05-20 by AuroraWright

MasterGamingYT, 2019-04-06:
I have a question for you Microsoft........
Which word is more offensive "Luma" or "3DS" because you probably think that luma3ds is a virus, but in reality, it isn't ¯\_(ツ)_/¯