LumaTeam/Luma3DS #458: [Suggestion] 3D Splash Screen

kehkou, 2017-05-21:
Would it be at all possible to update Luma to have the option to use a 3D image for the top splash screen? 
Or is it something that is not possible?
Thank you!

jonirrings, 2017-05-22:
I suppose not. luma boots before the system load, and 3D is supported by the system. just guess

kehkou, 2017-05-23:
That is exactly why I think it could not be possible, but I am also guessing.

As for the formatting, it is consistent with the formatting on other [Suggestion] "issue" topics.

hartmannaf, 2017-05-23:
It could be possible, but someone would need to find out how to set up the screen that way.

Kaisogen, 2017-05-24:
How about a menuhax like solution? Is there anything like system.xml on the Wii U on the 3DS we can edit?

This would drastically increase the amount of time Luma takes to boot, but, it could go like this.

>Install process or application

>Luma turns on. 

>Goes through regular boot 

>Automatically open application or process

>Display image for 5-3 seconds in 3D. Disable home button and no manual break so you can't exit early

> Go to home.

Something like this could work, assuming we have the technology to do it. I don't really know though.

WubTheGame, 2017-06-10:
can we at least have animated boot screens?

Kaisogen, 2017-06-11:
I believe BootCTR9 has support for this.

WubTheGame, 2017-06-11:
i can't find a version of it for boot9strap though

hartmannaf, 2017-06-12:
@WubTheGame They are inside of the gbatemp release thread: https://gbatemp.net/threads/release-the-bootctr9-release-thread.435347/page-3#post-7370533

Hikari-chin, 2017-06-19:
Thanks. This will be implemented at some point in the future most likely, but it is not a high priority.

TuxSH, 2020-04-15:
Closing to clean the issue tracker up

closed 2020-04-15 by TuxSH