LumaTeam/Luma3DS #460: [Suggestion] Rendering (Smooth Edges) for DS and 3DS Games as B9S is capable of this.

Bolt23, 2017-05-22:
As we have a versatile exploit (B9S) it would be easy to code a payload which could smoothen the jagged edges for both DSi (TWL) and 3DS (CTR) Games it wont be a FPS degrading factor as puny classic emulators such as gbSP has this to smoothen the edges of GBA games which were like crap before. 

The whole 3D Shacking Community are your Debtors @AuroraWright

piratesephiroth, 2017-05-22:
You want to smoothen the jagged edges on 240p screens? OK, yeah that would sure make a ton of difference...

TuxSH, 2017-05-22:
> as B9S is capable to do this

No.
Just no.

closed 2017-05-22 by TuxSH

Shadow322, 2017-05-22:
Ikr nobs need to use the system like it was designed

Bolt23, 2017-05-23:
@TuxSH Why so? Answer it! Will ya?

Margen67, 2017-05-23:
@Bolt23 B9S is no more capable of doing this than A9LH, Menuhax, MSET, or any other entrypoint.

Filtering only helps for DS and GBA games because they're a lower resolution.
If you were to apply *blur* to something running at native resolution (3DS games) it would be just that: **Blur**.
What you really want is Anti Aliasing which can smooth out jaggies, but isn't possible on the 3DS due to technical limitations. Certain games can't even run at full speed normally.
Not to mention like piratesephiroth said the resolution is too low for it to make a difference regardless.

Your best bet for running 3DS games at a higher resolution is with the emulator Citra.