LumaTeam/Luma3DS #461: Out of region DLC not working

AkitoCL, 2017-05-22:
Back in A9LH out of region custom DLC worked perfectly, specifically with the latest Taiko game which required a .txt file named as the title id and having JAP JP written on it on the locale folder.

Now in b9s it doesn't work even when checking the Enable Game Patching option and having the .txt in "locale"

AuroraWright, 2017-05-23:
The issue template requires you to read the wiki. The wiki has the solution for this

closed 2017-05-23 by AuroraWright

AkitoCL, 2017-05-23:
Except the wiki doesn't have a solution. I have the right text file in
locale and it still doesn't read the DLC, before it did. This all happened
after I was done with b9s.

cheatfreak47, 2017-05-23:
@AkitoCL Generally when the _author of the software you are using is telling you something you are doing is incorrect, they aren't lying._

jonirrings, 2017-05-23:
@AkitoCL 
First, pls use the issue template. github is not a forum
Second, show the path you locale.txt file and the region of your game app and its dlc.

Read the wiki, and go to `Enable game patcher` part, you will see that **THERE IS NO LOCALE FOLDER AT ALL**. so maybe you configured a wrong locale path.

AkitoCL, 2017-05-23:
You people... were actually right. I didn't read right and in my rush somehow I read the old instructions for locale files.
My mistake, all of this.