LumaTeam/Luma3DS #462: [Suggestion] Privacy options

Poliwrath, 2017-05-22:
Like, a option that disables most online connectivity features (MiiVerse, "Friends", StreetPass, SpotPass) while still keeping the Wireless tab on (for things like multiplayer games or homebrew apps that use the internets)

TuxSH, 2017-05-23:
Spot-Pass, Street-Pass: you can disable it per-game

Other features: use a custom DNS server.

closed 2017-05-23 by TuxSH

Cottoneyes, 2017-05-26:
So we should probably reopen this, huh?

arbingordon, 2017-05-26:
Why should it be reopened when Tux gave you a complete solution?  Too lazy?

HoLLy-HaCKeR, 2017-05-26:
Building this into the CFW would be a way more convenient solution, and doesn't require running a DNS server and the privacy/security issues that could come with it.

TLATER, 2017-05-26:
Reasons to bake this into the CFW:

- Some people won't realize this is a problem, resulting in more future banwaves. This makes hacking less safe and is an issue for the community.
- Data can be sent before users have a chance to turn the services off, since changing network settings requires turning on network connectivity.
- Even if they realize this can happen, it's hard for users to find exactly all settings they need to turn off. A simple oversight can cause a ban.
- This would not require users to use an external service to create a DNS server, which makes things easier for less tech-savvy users.
- The convenience of a simple switch would make more users do this, making hacking overall safer.

Besides, we know Nintendo is after us now, it's time to start the arms war.

urherenow, 2017-05-26:
If your router doesn't have a switch, pull the plug. Done.

If you want online, fix your settings then turn your router back on.

TLATER, 2017-05-26:
@urherenow Not all users can do that. Some may be on public networks, some may require the connection for other devices. You *can* always change your 3ds settings somewhere you don't have a network connection setup, but that still seems stupid.

This also completely ignores the fact that some users simply don't realize this could be a problem.

TuxSH, 2017-05-26:
http://gbatemp.net/threads/poll-bans-and-send-information-about-the-console-to-nintendo.471814/

The poll is conclusive about the fact that half of the people who had this opt-out option enabled were banned (the other banned people had probably the option enabled at some point).

I don't have to provide such an option, _as Nintendo already did_ (System Settings > SpotPass > Send information about installed titles (or something like that)).

Moreover if it's "baked into CFW" Nintendo could find a worse way to detect and ban consoles. I'd rather not have that happen.

TLATER, 2017-05-26:
@TuxSH 

> Moreover if it's "baked into CFW" Nintendo could find a worse way to detect and ban consoles. I'd rather not have that happen.

Wouldn't they be able to find such a worse way with the current "fix"? It should be easy to pull a Microsoft with the next firmware update and simply send data regardless of that setting. If they wait a few weeks to trigger ban waves until everybody has updated, they could seriously weed through the community with such a strategy.

It seems *safer* to let a third party prevent sending data, rather than leaving this to Nintendo, who have an incentive not to prevent this.

TuxSH, 2017-05-26:
It's playing cat and mouse and it'd be better if Nintendo doesn't do more drastic checks.

Kaisogen, 2017-05-26:
I don't even really use the eShop or online often, as I was banned beforehand (for reasons I'm not listing due to legality)

I think it would be interesting if we could see where the telemetry was being sent to, so we could block those routes.

Plus, humans make mistakes. Sometimes I accidentally click yes on spotpass.

Storm-Eagle20, 2017-05-26:
> It's playing cat and mouse and it'd be better if Nintendo doesn't do more drastic checks.

What sort of checks could they possibly resort to that wouldn't result in false bans? Worst case scenario we will be forced to go back to RedNANDs.

justinjustin7, 2017-05-27:
https://gbatemp.net/threads/regarding-the-recent-3ds-banwave.471781/page-106#post-7337688

From what I'm understanding from PF2M's post is that Nintendo is collecting this data regardless of the user settings. Couldn't someone look into patching the BOSS applet to remove data on anything hacking related before it gets sent?

bakugo, 2017-05-27:
Honestly you people should just let it go.  
Luma3ds devs are too busy acting morally superior to add any useful features.
This exact thing happened back during the Pokemon bans and they said the same.

ShadowEO, 2017-05-27:
@bakugo It is not the same thing that happened in the Pokemon bans, the Pokemon bans occurred because people connected to Nintendo's Online Services with a game that was not released at that time. It had nothing to do with the Luma devs, or CFW of any kind. Only that pirates and people who were unfortunate enough to get a copy early went online before a game's release and were banned for it.

mercenaryship, 2017-05-27:
ALL RIGHTS RESERVED

bakugo, 2017-05-27:
@ShadowEO It's the same thing. Both bans were attempts to punish piracy. My point is, Luma3DS devs (and the hacking community in general) act like they're friends with Nintendo and don't want to get on their bad side, which makes absolutely no sense because 1. 99% of people with CFW use it for piracy (don't bother denying it because we both know it's true) and 2. installing any kind of CFW or homebrew is against Nintendo's terms of service, even if you don't use it for piracy, so you're already on their bad side no matter what. Refusing to add useful privacy features because "nintendo might come after us" is therefore a nonsensical decision.

urherenow, 2017-05-27:
Look... This isn't a discussion forum. It's a place to submit BUGS and fixes for those bugs. This is open source, and all are welcome to fork, clone, and modify they see fit.

In other words, do it yourself and submit a pull request. Otherwise learn to take a polite hint, before it's time to not be so polite.

Storm-Eagle20, 2017-05-27:
> Look... This isn't a discussion forum. It's a place to submit BUGS and fixes for those bugs.

It's an 'issues' section and most developers, notably Plailect, are quite open to having suggestions reported here. And this is AuroraWright's project, not yours, so AuroraWright (and TuxSH?) makes the decisions.

> This is open source, and all are welcome to fork, clone, and modify they see fit.

> In other words, do it yourself and submit a pull request. Otherwise learn to take a polite hint, before it's time to not be so polite.

Most people would do take a crack at it if they knew how to even code it. This also raises the issue that almost no one really knows about the BOSS applet other than AuroraWright and a select few people.

urherenow, 2017-05-27:
You are completely missing motre than one point.

1) TuxSH HAS already chimed in. More than once.

2) despite the fact that he has, in the most polite way possible, dismissed this suggestion, you and a couple more people continue to chime in like this is a forum. It's not.

It's long past time for you all to fish for talent in temp or reddit, or irc.

Demirramon, 2017-05-27:
There's way too many reasons to do this, I don't get any of you saying this is not necessary, **because it is**.
Wasn't the idea of CFW to be always able to use homebrew safely? Who is going to keep using it if you get instantly banned when using it if you don't do a bunch of extra steps that, anyways, won't guarantee your safety?
I don't know how complicated would that be, but it needs to be done. Craziest stuff has been done.

urherenow, 2017-05-27:
Then do it akready. Or shut up and go home.

urherenow, 2017-05-27:
*already

(Time to NOT be polite anymore)

jordguitar, 2017-05-27:
Yes, let's just insult people because they do not have the same view as you. WORKS EVERY TIME!

This is a legitimate concern now. Eventually, the reason behind the ban will be found out and there needs to be some action taken.

ForcedToRock, 2017-05-28:
I want to add that I was one of the very few that had Spotpass off, and still got banned. The point was made about it being enabled at some point. Well, the latest possible point was February when I did a NAND restore, I think I ended up having to format the system. Every time after setup, I make sure to disable Spotpass and the automatic downloading of software, because those sudden "new software" notifications bug me. I've done this since I first installed A9LH in early 2016. AuroraWright posted poll after this that included the Friends List setting of showing titles being played. I lean towards that being a culprit because that was foolishly left enabled by me, quite honestly I thought I had it disabled. 

If the suggestions by the OP can't be baked into the CFW, then how about a well-written tutorial on how to block the sites the 3DS sends data to ourselves? According to this post, there's a bunch.

https://gbatemp.net/threads/guide-how-to-block-fw-updates-via-windows-7-8-8-1-virtual-hotspot.386177/

At this point damage has been done, I don't expect any dev to provide a magical solution.

mercenaryship, 2017-05-28:
> via-windows-7-8-8-1-virtual-hotspot.386177/

Rocknrolla? The princess goes in the sanctuary first.

ALL RIGHTS RESERVED

Margen67, 2017-05-28:
@mercenaryship wut

techmouse, 2017-05-28:
The message Nintendo has sent with the most recent banwave is clear. If we use **any** homebrew, not just CFW, we are their enemy. We can no longer lie to ourselves or each other. As much as we want to co-exist with them, they won't allow it.

Our only options are obvious. We can either stop leaving weak spots open, or we can be at their mercy time after time.

Monotek18, 2017-05-28:
Don't want to be that guy, but if teh data aren't sent while they must be sent, Nintendo will know it.

Sadly I don't see a smart way to block ban that could work :/

cheatfreak47, 2017-05-28:
@Monotek18 Send nintendo _false_ data that looks legitimate. 
I bet I just totally blew your mind just now.

Monotek18, 2017-05-28:
@cheatfreak47 Yes of course you can set it offline, and senting false data would be the best way, but not sure if it would be 100% safe.

And nope you didn't blow my mind.

Eric-Jackson, 2017-05-28:
RE: Privacy options

Instead of disabling the sending, could any unofficial apps be logged as official ones? I don't think there's any negative to that as far as Nintendo being able to change the way they're detecting hacks. Seems like a QOL improvement to me.

TuxSH, 2017-05-28:
So you're reposting comments from the other issue, heh.

I've explained my point of view in #467.

I'm locking this issue for the same reason as #467.