LumaTeam/Luma3DS #463: outdated info/missing parts

NWPlayer123, 2017-05-23:
I'm sure you're aware, but currently we need to download firmtool separately and get the Makefile to read it, so adding it as a dependency like GodMode9 has already done is ideal, otherwise you no longer need --recursive in the README cause all the other external repos have been ditched for pure boot9strap.
If anyone stumbles upon this, I've been pasting firmtool.py in the root directory here and editing the Makefile (line 68) to add the .py so it finds it

urherenow, 2017-05-23:
It might help if you actually installed firmtool. It happens to install to python as a module, same as the python crypto.

NWPlayer123, 2017-05-23:
I'll have to try that, still should work out of the box without doing that tho

dragos240, 2017-05-23:
Here's another one I found:

**Explanation of the issue:**
The "I get an auto-shutdown with GBA/DS/DSiWare games" in the "FAQ and Troubleshooting" section of the wiki still links to [The A9LH section (which no longer exists) of Plailect's guide](https://plailect.github.io/Guide/installing-arm9loaderhax).

**Steps to reproduce:**

1. Visit the Wiki, then the FAQ and scroll down to "I get an auto-shutdown with GBA/DS/DSiWare games". There's a broken link to the [A9LH section of Plailect's guide](https://plailect.github.io/Guide/installing-arm9loaderhax).

urherenow, 2017-05-23:
@NWPlayer123 looks like you are correct. Running the install script for firmwaretool seems to install firmwaretool.py into (on my windows setup with python 2.7.x) 

c:\Python27\Scripts\  (the trailing "\" is important in your environment path)

So although you do not need to copy firmwaretool.py anywhere, you DO have to edit the makefile line 68 to add .py to it. I'm guessing that the firmwaretool (without .py) is for *nix setups only or perhaps it will install to python and run correctly as-is if run from msys instead of a regular command prompt (can't be bothered to try).

TuxSH, 2017-05-23:
It's firmtool.

I've edited the README a bit, tell me what you think of the changes.

urherenow, 2017-05-23:
@TuxSH still have to use firmtool.py. I updated ctrulib, and I updated and re-ran your firmtool install script. What gets installed to Python is firmtool.py, not firmtool. Simply typing firmtool does nothing on windows (usually).

If you're on Windows and only need to type "firmtool", can you show me your registry entry for .py files that makes it automatically recognized as an executable?

TuxSH, 2017-05-23:
Isn't that set by the python installer, at least with recent ones?

Class .py default key is Python.File ...

NWPlayer123, 2017-05-24:
yeah, uh, installing firmtool only copies firmtool.py to C:\Python**\Scripts (**=ver), I copied both from the repo and now it's happy with just "firmtool" (cause the file exists in my path), that should probably be fixed on that side tho

urherenow, 2017-05-24:
the above works for me, only in msys. regular command prompt still won't take "firmtool". Either manually copy firmtool and use msys, or change makefile to use firmtool.py. It works, but not as-is.

urherenow, 2017-05-26:
interesting... running "pip install --upgrade git+https://github.com/TuxSH/firmtool.git" from msys resolves everything and "firmtool" works perfectly. Guess I need to remember to use msys from now on for luma.

piratesephiroth, 2017-05-28:
"firmtool" never works here either.
I must use "firmtool.py"

urherenow, 2017-05-28:
Damn, son... my post right above tours tells you what to do. Install firmtool from within msys prompt (not cmd.exe). Then it will run fine from msys. It will NEVER run from CMD.exe because .exe, .bat, and .com are the only extentions you can ignore in windows.

piratesephiroth, 2017-05-29:
msys? If you mean I should run from sh then nope, it doesn't work either.

[image]

Jerry-Shaw, 2017-05-29:
@piratesephiroth We are at the same place. The only way to fix this is to place the "firmtool" file right in the Luma folder. I tried to install firmtool and confirmed that "firmtool" command ran, but when building Luma, the command was not found still. Very strange.
I am under Windows 10, using devkitARM, msys is in devkitARM.

piratesephiroth, 2017-05-29:
I think the easiest fix is to change "firmtool" to "firmtool.py" in the makefile.
Would that break anything in a non-windows OS?

urherenow, 2017-05-29:
I mean msys. If you installed devkitpro (vice just the devkitarm installer) you have devkitpro/msys/msys.bat

Install and run through that. But SH... are you not not on Windows?

Margen67, 2017-05-30:
Issue template is outdated as well.

TuxSH, 2017-06-01:
latest firmtool should fix the .py issue.

closed 2017-06-05 by TuxSH