LumaTeam/Luma3DS #464: [SUGESTION] System-wide keys modifier / hotkeys

jokkovitch, 2017-05-25:
A lot of users got their home button damaged / not working.
Would it be possible to create hotkeys, like L+R+UP+X would emulate Home button, system-wide and from boot.
Painfull not being able to enter God9 options just because of a malfunctionning button.
Something like this was created for Jailbroken Iphone and litteraly tons of people were using it.

Kaisogen, 2017-05-25:
I'm pretty sure there was a way to send inputs to a computer through the 3DS. Then send the inputs back to the 3DS as something else.

This could be done with a PC program. I don't know if it would be possible to do this through Luma.

deadair3210, 2017-05-25:
This would not work in GodMode9 as far as I know even if it is possible, due to GodMode9 loading before Luma

⁣Sent from BlueMail ​

urherenow, 2017-05-25:
Umm... It doesn't HAVE to load before luma. GM9 can run from HBC, and some direct entry points as well.

dogtopus, 2017-05-25:
Yet another reason to have a background thread I guess.

TuxSH, 2017-05-26:
"Painfull not being able to enter GodMode9 options just because of a malfunctionning button."

GodMode9 is running in baremetal (w/o an OS), what you describe is a GM9 issue more than a Luma3DS one.

closed 2017-05-26 by TuxSH