LumaTeam/Luma3DS #467: 3DS banwave

Derek1199, 2017-05-27:
I understand there is probably no way to fix this at all except for disabiling sending of information and hiding what title is currently being played in the friends list applet. But we need a way to patch it so nintendo can't ban people like that. The cause of the ban is the friends list applet. Basically what is happening is a invalid title ID is seen, then a ban request is sent, then the user's 3ds gets banned from online play. The only way to unban is to inject a not banned localfriendseed_B or sell themselves to hackerone. This is a very serious issue. Affecting all cfw users.

Storm-Eagle20, 2017-05-27:
I don't understand why people are so openly against patching the BOSS applet especially considering that's the biggest part of our problem. Have an upvote, this isn't something that should be ignored.

Monotek18, 2017-05-27:
if you block the TID or other information from being sent, Nintendo will know.

Demirramon, 2017-05-27:
If Nintendo is able to know we are hiding stuff, we can always mask it, right? Like, displaying a legit title instead of the one we are using. We would still be sending data to them, just not real data, so I don't think it would be suspicious.

Monotek18, 2017-05-27:
The problem is that if you change an homebrew TID to a legit title, well you can't have 2 same title ID installed, and you'll have to do it manually definitly, and if the CFW would be able to change illegal TID to legit TID it will take much more time to boot, so not the good way.

I still really wonder how they could make a CFW that won't be the target of Nintendo, and since it's open source Nintendo can check how it work so easier to target the users.

seiya-git, 2017-05-28:
#462

jaxrogers2, 2017-05-28:
I want to note that for the friends list applet, there is a way for it to say "the title being played cannot be determined." If a flag could be set where any TitleID not on a whitelist of official game TitleIDs shows up as "the title being played cannot be determined,"  this could prevent this data from being harmful (or this sent data to be useless).

Also, for those concerned about the project being open source, if someone wanted to implement this they could probably (with permission) fork the project but then only post releases (like NTR does).

urherenow, 2017-05-28:
This is github, not a forum. You don't need permission to fork ANYTHING here. Wish you needed permission to post issues though...

jaxrogers2, 2017-05-28:
Well, you would need permission to fork the project without posting the source (it believe it would be breaking the GPL). But you're right that anyone can fork the project. It's just that if we want to implement these anti-ban measures, we would need the code to be closed source so Ninty can't just look in the source to bypass our efforts, or even worse, ban those matching the bypass.

TuxSH, 2017-05-28:
> so Ninty can't just look in the source to bypass our efforts

You know, Nintendo could check your account to check your account to check if you actually owned the game you're connecting with. They don't seem to do that _yet_. (don't get me wrong on there, I don't condone piracy **at all** but I'm very well aware most of you guys don't own all the games you're playing to...)

Right now you can avoid being flagged by disabling those telemetry options. This only takes you one minute to do so, and is very straightforward.

Tsume, 2017-05-28:
You say that while not knowing if that's the actual cause. Even further by outright ignoring the fact that people who've had those features off for years still getting banned. Willingly ignoring information doesn't make your state anymore correct/incorrect. Just makes you look like you're intentionally ignoring the issue. Which does no one any good. Having those disabled doesn't keep you safe. I know this to be fact, because my o3ds that hasn't been turned on in 2 years(both of which were disabled the moment I got it). Was still banned.

TuxSH, 2017-05-28:
> You say that while not knowing if that's the actual cause.

Not with entire certainty but polls and other research indicate it's almost certain.
Then why are you asking for it anyways?

> My o3ds that hasn't been turned on in nearly 2 years

It could have been banned within those 2 years. Accepting the ToS again also reenables the option fwiw.

cheatfreak47, 2017-05-28:
I'd imagine if this were done it'd be done well enough that Nintendo, source or not, wouldn't be able to pick up on it very easily, the 3DS isn't an "always online" console, if we suddenly cut off their supply of data, they would probably assume the system in question is offline, and even if that were true, we could also just _spoof_ data that looks legitimate.

@TuxSH Piracy or not, Nintendo has set a clear precedent here. They don't care if you're a pirate or a homebrew user, they will ban you for _any_ unauthorized software and that's that.

You fellows are the people behind the biggest CFW out there and people running it are getting _permanently banned from online play._ It really shouldn't be rocket science to realize _why_ people are asking for a function like this. CFW, being that it can patch just about anything running on the system, _including BOSS_ is the most obvious vector for mitigation of a massive effort to ban every last user running something big papa Nintendo doesn't like.

If you really believe disabling those options is effective, then you could just add a patch that forcibly disables those settings. Luma is supposed to be n00b proof, right?

And don't get me wrong here, I have a ton of respect for you guys, and whatever you decide to do is fine, even if thats nothing at all- but there's my two cents.

urherenow, 2017-05-28:
@TuxSH there is no way to tell legitimacy of a cart. Nothing prevents loaning one out. In that respect, looks like Gateway and Sky wins (provided one uses a unique and valid ID. But online never worked anyway, without an ID that was valid. Bans came from multiple people using public IDs online simultaneously...)

But anyway, these kids are sucking you into using this as a forum (guess I'm guilty too now)

TuxSH, 2017-05-28:
Making an easy or prominent way to circumevent that **will** make Nintendo look into more effective, purely server-side, ways to ban (moreover the current situation could be worse, they let you disable what's leading to bans...)

Remember this PR: https://github.com/AuroraWright/Luma3DS/pull/286. Nintendo fixed that on their end even if it was never merged.

With all due respect, I found it astoundishing that you guys fail to realize this.

PS:
> there is no way to tell legitimacy of a cart.

That's the purpose of private haders.

urherenow, 2017-05-28:
?

Did we post at the same time or did you have time to quote me? Either way, you dump 1 private header of your own, and nothing stops you free m using that same header on every downloaded rom dump (at the moment, anyway).

Monotek18, 2017-05-28:
you can set the same private header to all your .3ds rom and it'll be fine.

urherenow, 2017-05-28:
Damn, I hate typing on my phone. No edit button in Safari to correct typos

TuxSH, 2017-05-28:
> Did we post at the same time or did you have time to quote me? 

Github automatically refreshes posts.

Whatever, not the subject.

cheatfreak47, 2017-05-28:
@TuxSH Correct me if I'm wrong but wasn't #286 a patch that bypassed an existing ban, as opposed to preventing people from getting banned in the first place?

If so, I'd say that hardly sets any kind of precedent for what your actions here should be.
The focus here should be on using CFW to prevent people from being banned, not unbanning existing banned people.

TuxSH, 2017-05-28:
My point is that when Nintendo wants to do something they'll have it done.

They will obviously fix any publically known workaround and make the checks even more severe.

cheatfreak47, 2017-05-28:
Well, considering people are already "disabling their only method for checking if they should ban you or not" en masse, I don't think adding a patch that sends them less information or even just disables those things by default is going to make them do anything they aren't already going to do.

It's also easy to figure out if someone that shouldn't be playing is playing, it's a lot harder to figure out _if_ someone should or shouldn't be playing in the first place, especially if they don't have a lot of information to work off of except a _lack of information._ 

The notion that we shouldn't do anything out of fear for something even worse happening is really silly, logically. We're already at the bottom as it is, we have nowhere to go but up. Their checking is already so severe that the entire community is _still_ in shock over the situation.

On top of that if they _do_ make checking more severe, they have to blow a lot of time and money on it to prevent it from false-positiving people, we have the upper hand here in that we're already at the bottom, what more could they do? Brick us?

I hardly think Nintendo could or would stoop that low, nor could they legally in some parts of the world.

AuroraWright, 2017-05-28:
If they feel like spending money on it they could add CFW specific checks (at the moment, there probably aren't any such checks)

jaxrogers2, 2017-05-28:
Yeah all of this banning is still guessing with BOSS data on their part. They don't truly have a check for CFW itself- it's more that they have information that identifies the symptoms of having CFW installed and then bans based off of that.

cheatfreak47, 2017-05-28:
@AuroraWright Well, to be frank, I'm pretty sure if they were going to do that, they either would have by now, or decided against it because they know we could just patch those checks out.

jaxrogers2, 2017-05-28:
Checking for unlegit titles is easier anyway and gets consistent results for bans compared to if a CFW specific check was developed that could be easily patched out or worked around.

mercenaryship, 2017-05-28:
Matt,

Is this Nintendo USA or Japan doing the banning?

ALL RIGHTS RESERVED

Rajh, 2017-05-28:
Hi, wouldnt it be possible to split the system in two parts ?

The first one with all legit stuff no homebrew no cia only a backdoor to launch the cwf. Then the cwf with the possibility to spoof (by choice) the LFS with a public one (probably banned but whatever). 

Meaning it would be possible to play legit game online without fear ban. And do homebrew stuff without care about online.

jordguitar, 2017-05-28:
To do that you would have to, in theory, make a hypervisor to do that.

techmouse, 2017-05-28:
>@TuxSH Piracy or not, Nintendo has set a clear precedent here. They don't care if you're a pirate or a homebrew user, they will ban you for any unauthorized software and that's that.

@cheatfreak47 is right. This banwave is **not** an attack on piracy. It's an attack on homebrew. We can't keep burying our heads in the sand on this one. As long as we have homebrew, we are their enemy.

mercenaryship, 2017-05-28:
What do you want to "fix"? This thread was unsubscribe or muted yet still shows in my inbox. Is there something your u want?

TuxSH, 2017-05-28:
> This banwave is not an attack on piracy.

Hard to tell when most people unfortunately use CFW for piracy.

closed 2017-05-28 by TuxSH

Plailect, 2017-05-29:
It seems a bit ridiculous to me that anyone here feels they have the right to *demand* that @TuxSH or @AuroraWright implement a feature *in their free time*. This project is maintained by them for no other reason than because they want to, and they have no obligation to implement anything they feel has too little gain to be worth the time it would take to implement.

It does not matter whether or not we as a community agree with their decision. If anyone here feels that patching BOSS for this feature is worth someone else's time and effort, then surely they can feel that it is worth their own time and effort to fork, implement, and pull request it themselves.