LumaTeam/Luma3DS #469: Accidentally updated N3DS to latest Luma hourly firm payload, now won't boot properly

Peacock1090, 2017-05-27:
I have a N3DS that was on 11.2. It had CFW with A9LH, until Sighax was announced and I followed the directions on 3ds.guide to change all the Luma payloads to .firm.

Things were working just fine for a while. Unfortunately Luma is weird with my N3DS with my cia of Pokemon Moon, so I tried to get to see if I can restore it to the older version so I could boot it up without the typical Nintendo error message. But whoops-a-daisy I thought updating Luma to the hourly was the best idea for whatever reason and now all I get is a blue light coming on for half a second before it shuts down again. Again, other than the weird error message with Pokemon Moon that I knew at least had a temporary fix, it was working just fine, though I later heard there were issues with the hourly update.

Is there a way to recover from this? From what I read it's TECHNICALLY not a brick but I can't boot up into anything with this going on. Not even by using the payloads like start_GodMode9 or anything. If I'm screwed for the moment, will there be a fix in the future regarding this? I'm honestly frustrated that my console can still technically work but something that apparently I'm not getting is just not allowing it to work or letting me even find what could be causing this error.

KunoichiZ, 2017-05-27:
Restore your boot.firm on the root of your SD card.

Peacock1090, 2017-05-28:
Welp, I feel dumb. Just did that and it worked just fine.

Thanks a bunch, guess I just need to avoid the hourlies for a while.

closed 2017-05-28 by Peacock1090