LumaTeam/Luma3DS #470: Feature Request: Hide EmuNAND options in config if SD Card isn't formatted for EmuNAND

sora10pls, 2017-05-28:
If it's possible to tell how the SD Card is formatted, then perhaps Luma's config could hide all EmuNAND-related configuration options if the SD Card isn't formatted for EmuNAND.

Dysproh, 2017-05-29:
Luma can detect if an emunand is present, so I dont see why this can't be a thing.

Kaisogen, 2017-05-30:
I personally find it fine. Some options are already removed upon SDless boot. This is a QoL change that I find nice, but isn't really necessary.

HugaShire, 2017-06-22:
If you can build Luma3DS, you can also do what I did. In the Config.c, find every line that says =isSDmode and change it to false.

quantumcat1, 2017-08-10:
Along the same lines, can there be a notice in the config if it has been booted from CTR-NAND? Lots of people have troubleshooting difficulties when it turns out that Luma was booting from CTR-NAND not SD card like they were expecting.

sora10pls, 2017-08-10:
I second what @quantumcat1 requested. That would be incredibly helpful.

fryfrog, 2017-09-12:
@quantumcat1, you should make this into an issue/feature request.

closed 2018-08-21 by AuroraWright