LumaTeam/Luma3DS #474: [Feature Request] Patch out signature checks of certificate chain

neobrain, 2017-05-29:
**Explanation of the issue:**

Luma currently does not patch out signature checks for the certificate chain in a CIA file. This makes it hard to write CIA generating tools easily, since that currently requires the copy of a (arguably copyrighted) valid signature.

Ideally, Luma should just patch out the certificate chain signature checks in addition to the signature check patches that are already in place.

**Steps to reproduce:**

1. Take any currently installable CIA file, change a byte at offset 0x2044 (which is the signature of the first certificate in the chain)

2. Try installing the modified CIA using e.g. FBI and observe it failing to do so even though we merely modified a signature

neobrain, 2017-05-29:
For reference, TuxSH suggested on IRC to try the following:
> 22:27 < TuxSH> there's a thumb function in p9 starting with FF B5 FF B0 15 00 AE B0  0E 00 28 00
> 22:27 < TuxSH> could you stub it out?

Will be busy for the next couple of days and I'm not exactly sure where/how new patches would be added, so it'll be a while until I can actually try it out.

TuxSH, 2020-04-15:
Closing, tracker cleanup

closed 2020-04-15 by TuxSH