LumaTeam/Luma3DS #480: "Show NAND or user string in System Settings"

Shadow322, 2017-06-01:
Does not work from nand aka rw/luma/customversion_sys.txt

sora10pls, 2017-06-01:
Works fine for me...

Margen67, 2017-06-01:
@sora10pls With customversion_sys.txt on NAND or SD?

sora10pls, 2017-06-01:
Works fine with CTRNAND (SD Card not inserted and in the path mentioned by OP), as well as with SD Card inserted.

Shadow322, 2017-06-02:
customversion_sys.txt with top line being Luma3DS-%d.%d.%d-%d does not work and yes its enabled and only shows Sys 11.4.0-37U on Luma3DS v7.1-8308e1a8 from NAND

sora10pls, 2017-06-02:
Aren't the latest nightlies broken? Your commit string displays latest commit, I'm running stable v7.1.

Shadow322, 2017-06-02:
hearing and doing is 2 different things

sora10pls, 2017-06-02:
ok

Shadow322, 2017-06-02:
Just tested with v7.1 release does not work either so I was right it should be fixed for the ones that it works for is because u have it in your Luma folder on your sd card

Margen67, 2017-06-02:
Luma will only read/write where the payload (boot.firm) is.

If the custom version string isn't in the same directory (SD card/CTRNAND) it won't read it.

Shadow322, 2017-06-02:
so it needs not be in the luma folder and be in root of nand?

Shadow322, 2017-06-02:
nope still does not work

Shadow322, 2017-06-02:
because its in rw/luma/ like with godmode9.firm the config.bin as well as the 2 splashscreen bins and all the other stuff works

Margen67, 2017-06-02:
No. If the luma payload (boot.firm) is on the SD card, the custom version string should be put in SD:/luma

If it's on the CTRNAND it should be in CTRNAND:/rw/luma/

sora10pls, 2017-06-02:
>because u have it in your Luma folder on your sd card

What part of my
>Works fine with CTRNAND

Didn't you get

Shadow322, 2017-06-02:
it is in  CTRNAND:/rw/luma/ and does not work there is no boot.firm on sd card

sora10pls, 2017-06-02:
Tried out setting my `customversion_sys.txt` to be the same as yours (line reads `Luma3DS-%d.%d.%d-%d` as you mentioned), works fine in both SD Card boot as well as CTRNAND boot.

I can't replicate this as hard as I try.

sora10pls, 2017-06-02:
I want to reassure to you something that your last comment may have mentioned.

If boot.firm is found on CTRNAND, it loads that, even with SD Card inserted (and if it has boot.firm).

If boot.firm is **not** found on CTRNAND, but it is on SD Card, it loads that.

If the SD Card *isnt inserted* and you're running boot.firm from CTRNAND, it reads the "luma" folder from CTRNAND.

Likewise, if the SD Card *is inserted*, regardless of launch method, it reads the "luma" folder from the SD Card, *not* from CTRNAND.

sora10pls, 2017-06-04:
@Shadow322 any updates?

Shadow322, 2017-06-04:
still does not work

sora10pls, 2017-06-04:
Let me make sure. When booting, you:

- Have `boot.firm` on the root of CTRNAND?
- Have the SD Card **not** inserted when booting?
- Have `customversion_sys.txt` in `CTRNAND/rw/luma/`?
- Have "Show NAND or user string in System Settings" enabled on your CTRNAND's Luma config?

If the SD Card isn't inserted, it loads all files from CTRNAND.

If the SD Card is inserted, regardless of boot method (`boot.firm` on CTRNAND only, on SD only, on both), it will load files from SD Card rather than CTRNAND.

sora10pls, 2017-06-06:
@Shadow322

Shadow322, 2017-06-06:
Then that needs fixed

sora10pls, 2017-06-06:
@AuroraWright @TuxSH not sure if it's intentional or not, but refer to the last 2 sections of my above comment.

>If the SD Card isn't inserted, it loads all files from CTRNAND.

>If the SD Card is inserted, regardless of boot method (`boot.firm` on CTRNAND only, on SD only, on both), it will load files from SD Card rather than CTRNAND.

Shadow322, 2017-06-06:
OK when I put customversion_sys.txt in CTRNAND/rw/luma/ on both my O3DS and O3DS XL it works as expected and its not on the sd card at all but when i do the same on my N3DS XL it does not work as intended so its a N3DS/XL problem

closed 2017-06-19 by Hikari-chin