LumaTeam/Luma3DS #483: Luma does not boot on boot9strap v1.1 because of new agrc handling

qwertyatom100, 2017-06-04:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

7.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (4 )

Splash: ( Off)

PIN lock: (Off )

New 3DS CPU: (Off )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch SVC/service/archive/ARM9 access: ( )

Set developer UNITINFO: ( )

Enable exception handlers: (x)

--

**Explanation of the issue:**
Upon booting Luma on a 3DS with boot9strap v1.1 installed the following error message will occur:
An error has occurred:

Unsupported launcher (argc = 0).

NOTE: This issue is fixed in branch b9s-next

**Steps to reproduce:**

1. Upgrade your 3DS to boot9strap v1.1.

2. Attempt to boot.

**Dump file:**
N/A

sora10pls, 2017-06-04:
>NOTE: This issue is fixed in branch b9s-next

...

james-d-elliott, 2017-06-04:
^ This. They're working on supporting it.

qwertyatom100, 2017-06-04:
Just putting it out there.

sora10pls, 2017-06-04:
The issue is fixed though.

If someone doesn't want to encounter it, they should use B9S 1.0 with the latest stable release.
If someone doesn't want to encounter it, but wants B9S 1.1, then use any build from the b9s-next branch.

closed 2017-06-04 by AuroraWright