LumaTeam/Luma3DS #485: Cannot boot with boot9strap 1.2

ITJesse, 2017-06-04:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37J SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

7.1 stable and the last commit 4429cb2095c7294a97843433ae0259958b8a419c

**Luma3DS configuration/options:**

Default EmuNAND: (x )

Screen brightness: (4 )

Splash: ( x)

PIN lock: (x )

New 3DS CPU: (x )

--

Autoboot EmuNAND: (x )

Use EmuNAND FIRM if booting with R: (x )

Enable loading external FIRMs and modules: (x )

Use custom path: (x )

Enable game patching: (x )

Show NAND or user string in System Settings: (√ )

Show GBA boot screen in patched AGB_FIRM: (√ )

Patch SVC/service/archive/ARM9 access: (√ )

Set developer UNITINFO: (x )

Enable exception handlers: (x )

--

**Explanation of the issue:**

When I upgrade the boot9strap 1.2, the console cannot boot.
It shows:

> Unsupported launcher (argc = 0)

But I can boot the other payloads such as GodMode9.

**Steps to reproduce:**

1. Upgrade to the boot9strap 1.2

2. Power on the console

**Dump file:**

**Screenshot:**
[image]

Margen67, 2017-06-04:
#483 https://astronautlevel2.github.io/Luma3DS/ use the builds from the "b9s-next" branch

ITJesse, 2017-06-04:
Thanks, seems the last commit works.

ITJesse, 2017-06-04:
But the chainloader dosen't work anymore.

When I boot GodMode9:
[image]

[crash_dump_00000001.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/1050762/crash_dump_00000001.dmp.zip)

But I can boot GodMode9 as boot.firm without luma chainloader.

ITJesse, 2017-06-04:
I opened another issue.
#486

closed 2017-06-04 by ITJesse

MelonGx, 2017-06-05:
Use GM9 20170531 nightly.