LumaTeam/Luma3DS #489: [Question] Possible to store Rosalina menu combo in a config file?

sora10pls, 2017-06-05:
As the title says. Currently, it's only possible to change the combo during runtime, and gets reset to default (default being `(L) + (Down) + (Select)`) every time the device restarts.

blackwing182, 2017-06-05:
Plus to this.

I was going to report this minor issue right now.

Steveice10, 2017-06-05:
This would be especially useful to me as my D-Pad down button is broken. Having the homebrew-launching title saved to a configuration file as well would be helpful as well.

Milanpsmp, 2017-06-05:
I support this since my L button is almost broken

TuxSH, 2017-06-05:
In my todo list 😛

K0r0kS33D, 2017-06-06:
Glad to know its being worked on! Thanks so much.

Hikari-chin, 2017-12-10:
The combination can be changed by editing the config.bin now; no too currently exists to modify

closed 2018-01-02 by AuroraWright

deltaFigure, 2018-12-29:
Is this still being worked on, or is this finished?