LumaTeam/Luma3DS #493: An Exception Occurred when loading Smash for 3DS

DewGaming, 2017-06-05:
I am running B9S 1.2 and Luma 8.0. I am attempting to load up Smash for 3DS, but I keep getting a "An Exception Occurred" error.

Details: "Processor: ARM11 (core 0) Current Process: ctrapp (USA Smash Title ID)
You can find a dump in the following file: dumps/arm11/crash_dump_00000003.dmp (this is the fourth time loading it)"

Any fix for this at all?
P.S. I tried opening Pokemon Sun and that worked fin, so I don't know if it had something to due with the large processing power or what since that game worked but not Smash.  Also, I have attached the .dmp files that were made upon the exception: [Smash 4 Error Message Dumps.zip](https://github.com/AuroraWright/Luma3DS/files/1050968/Smash.4.Error.Message.Dumps.zip)

AnalogMan151, 2017-06-05:
Smash opens fine for me. More details?

New or old 3DS. Any romfs mods like music or stages? Any locale setting changes?

DewGaming, 2017-06-05:
Moving the code.bin file seemed to fix the issue.

ntojr123, 2017-06-06:
yeah me too im having the same problem opening things like crt quake and stuff like tthat but i dont know where to move code.bin or what it even is sorry im new to this please help im on luma cfw 8.0 right now and b9s 1.2

wkocbz, 2017-06-08:
Hi , i had the same problem, but i resolved it.
you open the luma menu ( start the console with power+select) and disable  gaming pathing.

ntojr123, 2017-06-08:
So I did that and it did not do anything now it says undefined instruction and then says you can find a dump in the following file dumps/arm11/crash_dump_00000042.dmp. right now I do not think luma 8.0 is stable so just try to wait for next update but any help would be appreciated.

TuxSH, 2017-06-09:
Try this build of the latest commit:
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1062691/boot.zip)

ntojr123, 2017-06-09:
How do I load that boot?

TuxSH, 2017-06-09:
you replace the current boot.firm at the root of your sd card?

DewGaming, 2017-06-09:
I tried using the provided boot.firm file in the zip file, but that file only leads to an infinite black screen.  I have the boot.firm on the sd card and have tried it was the boot.firm on the root, as well as a payload using Luma's chainloader.

ntojr123, 2017-06-09:
Ok so don't use that boot?

DewGaming, 2017-06-09:
I updated the original comment with more info, including where I got the code.bin as well as the settings I have for Luma and Smash-Selector (if it makes a difference)

ntojr123, 2017-06-09:
But where do I download the code.bin to fix it but anyways I'm about the Same settings two but with game patching​ disabled

DewGaming, 2017-06-09:
Read the "Steps to reproduce" section

ntojr123, 2017-06-09:
I did smash works fine for me but I'm talking about other applications but do I move the dump files or something

William-Imm, 2017-06-19:
Appear to be getting this very same error. Using Smash Selector 2.2. Tried using Luma 8.0 and nightlies, used code.bin from Smash Selector 2.2, nothing seems to be working for me.

[image]
[Crash dump file](https://www.dropbox.com/s/rjrfvvoxt8y2rpb/crash_dump_00000016.dmp?dl=0)

Hikari-chin, 2017-06-21:
Could anyone with this issue try a later commit like [this one](https://astronautlevel2.github.io/Luma3DS/builds/Luma3DS-1782827.zip)? Thanks.

DewGaming, 2017-06-21:
Still crashed the game, even on the latest nightly 0fbdeee3 that I got from https://jenkins.hakujou.fr/job/3DS/job/Luma3DS/

William-Imm, 2017-06-21:
@Hikari-chin I've tried a Luma nightly at the time to see if it would avert the crash, it still doesn't. In fact, I think it's the exact same one that you've linked here.

Hikari-chin, 2017-06-22:
Just to clarify, @DewGaming, the Smash mod you're trying to use is compatible with whatever update of Smash you have installed (or the original if none installed, etc.), right?

DewGaming, 2017-06-23:
Even without a saltysd folder on my sd card root, the game still crashes

William-Imm, 2017-06-24:
I've tried with the latest version of Smash Selector, using IPS patches. System still crashes, this time with a data abort.

thunderkai, 2017-06-26:
I have this issue too

angelmarlboro, 2017-07-01:
luma 8.1 dosnt correct this bug? you must selec in te luma boot menu Disable ARM11 exeption handlers

https://gbatemp.net/threads/2-issues-and-2-doubts-and-arm11-exception.475467/page-2

William-Imm, 2017-07-01:
Tried, the game just crashes directly to the 3ds home menu instead.

Hikari-chin, 2017-07-01:
@angelmarlboro @William-Imm Note that you forfeit your right to comment on GitHub issues if you choose this option, as specified in the config option description. Disabling this option prevents us from getting valuable crash information when issues happen.

ntojr123, 2017-07-01:
I'm quite confused about that because how in the hell are we supposed to tell them what's wrong

ntojr123, 2017-07-01:
For most of the apps it just brings me to the home menu and says needs to restart system

Hikari-chin, 2017-07-01:
Exactly, @ntojr123.  Keeping that option enabled allows us to get a  general idea of what went wrong. With that option disabled, there's no feasible way to trace where the crash came from. It's best to keep it enabled; the only situations I can see where it's viable to keep enabled is if you have a child or similar that is scared by all the numbers and letters they cannot understand, or if you're using an improperly made (buggy) homebrew or hack that causes crashes.

ntojr123, 2017-07-01:
I think that it should be different so that you can tell them what is wrong to try to fix them but yeah I understand you

William-Imm, 2017-07-01:
Understood, the option to disable the exception handler is disabled and will remain disabled. Only tried it out to see if it would work at all, and the answer is no.

[Attached relevant crash dump I've prepared earlier (data abort, using SaltySD IDS patches)](https://www.dropbox.com/s/60708pnldmm8msb/crash_dump_00000027.dmp?dl=0)

ntojr123, 2017-07-01:
So what do we do now?

Hikari-chin, 2017-07-01:
It's being looked into; that's why this issue is still open. Rest assured, I'll post a status update when something relating to it is found.

yamato05, 2017-07-06:
I also have the same problem with Hyrule Warrior Legends.
Adventure mode causes the exception but the Legend mode is fine.

https://www.dropbox.com/s/nox864gc5pphn52/crash_dump_00000016.dmp?dl=0

heyrict, 2017-07-07:
I had the same error after updating from luma6.6 to luma8.1 when starting sysnand, while emunand works fine for me. Should I open a new issue?

astronautlevel2, 2017-07-08:
Can we get a summary of the thread up until now to make testing easier?

LinkSoraZelda, 2017-07-09:
SaltySD requires Luma patching to run. It utilizes the LayeredFS/IPS patching abilities of the CFW.

If OP hasn't, they should try updating to Smash Selector 2.3, which uses IPS patches instead of code.bins. Then confirm that there is nothing other than the patch in luma/titles/(folder name with EDF at the and)/code.ips) Make sure absolutely nothing else is there as it may conflict.

BARNWEY, 2017-07-10:
@LinkSoraZelda The IPS patches cause even more issues, the older code.bin files work more often...

ewormPL, 2017-07-22:
Trying to run a Pokémon Y hack (code.bin), same issue.

Hikari-chin, 2017-08-27:
Going to go ahead and close this, given the recent Smash Selector release and mention of crash issues on their side.

closed 2017-08-27 by Hikari-chin

LinkSoraZelda, 2017-08-27:
Not quite. Smash Selector and SaltySD will only fail to load and close with an exception if the user is running a non-American copy of Smash 4 on patch 1.1.7.

This user seems to be running the American patch, and this issue was opened befote 1.1.7 released.

cetertel, 2017-09-11:
American copies of the game should be running perfectly fine, as nothing is needed after the 1.1.7 patch, like LinkSoraZelda said.

I am getting this same crash log and would be happy to give more crash dumps and pictures if you would like.

EnderWeeb132, 2018-03-01:
Currently having the same problem, and have had for a while. I have Smash 4 on cart, updated to 1.1.7. Gives the same crash. Seriously we need a solution for this

wazupttp90, 2018-04-02:
Have you guys tried taking the cia from the cart and installing as a eshop download

shinra358, 2019-03-05:
SAME WITH LUMA 9.1