LumaTeam/Luma3DS #495: [Feature Request/Issue] fix *hax payloads/*hax payloads crash

Zmxhawrhbg, 2017-06-05:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.3.0-36U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v8.0

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (1)

Splash: (Before)

PIN lock: (Off)

--

Autoboot EmuNAND: (Off)

Use EmuNAND FIRM if booting with R: (Off)

Enable loading external FIRMs and modules: (Off)

Enable game patching: (Off)

Show NAND or user string in System Settings: (On)

Show GBA boot screen in patched AGB_FIRM: (On)

Patch ARM9 access: (Off)

Set developer UNITINFO: (Off)

--

**Explanation of the issue:**

With the new homebrew launcher, you can't run anything that requires a target title (like JK's Save Manager or HANS)

However, trying to load the old homebrew launcher results in a crash, and if you use the homebrew launcher loader, then homebrew crashes on load.

Also, DSP Dump requires a hax 2.x payload, which doesn't work on Rosalina (it's completely possible that DSP Dump just needs to be updated)

**Steps to reproduce:**

1. Load the old homebrew launcher using homebrew_launcher loader

2. Load a title

3. CRASH

**OR**

1. Load the homebrew launcher using a payload like soundhax

2. CRASH

I've uploaded 4 dump files in arm11.zip

[arm11.zip](https://github.com/AuroraWright/Luma3DS/files/1051076/arm11.zip)

blackwing182, 2017-06-05:
All hax payloads are obsolete for now on.

Zmxhawrhbg, 2017-06-05:
I know that, but some homebrew rely on the hax payloads, and probably won't be updated for the Rosalina entrypoint.

fincs, 2017-06-05:
hax payloads by design cannot work under Rosalina because the memory map is shuffled around to make space for the Rosalina system module to load (and this breaks the hax payloads because they expect certain things to be at fixed locations in memory).

It was decided that target title selection is not useful anymore since 3dsx applications have full ARM11 system permissions now, and it is possible to directly access the savefiles of any title by using the appropriate FSUSER commands. In the meantime, you can use the CIA versions of savegame managers.

There should be a replacement for DSP Dump soon.

BigBrainAFK, 2017-06-05:
Use CIA versions of anything you can as they most often work way better (at least when +hax was a thing)

fincs, 2017-06-05:
With Rosalina 3dsx no longer has ARM11 FS permission or code size limits, so 3dsx running on Rosalina should be as good as installed applications.

closed 2017-06-06 by TuxSH