LumaTeam/Luma3DS #496: [Suggestion] option to disable rosalina menu in luma config

bladesdnm, 2017-06-05:
please give us the option to disable Rosalina menu.

I don't want to open it by accident, while in a game. or have my kids open it and mess around in it by accident. maybe add an option to disable it from the config screen. 

Thank you for your time.

sora10pls, 2017-06-05:
While the odds of you pressing (L) + (Down) + (Select) all at the same time seem extremely slim, this would be a nice option.

TuxSH, 2017-06-05:
Not possible.

closed 2017-06-05 by TuxSH

Gudf, 2017-06-05:
@bladesdnm Well, you can still change the combo by something with A+B+X+Y+Up+Right+L/R, which would really be nearly impossible to do unless you know how (and want) to do this.

TuxSH, 2017-06-06:
Oh, yeah, you can change the combo.

Valeri0p, 2017-06-06:
Why you can't?

bladesdnm, 2017-06-06:
the combo don't save/resets. why even have the option to change it if it don't work? 
why even suggest that as a solution u know damn well it don't work

Gudf, 2017-06-06:
#542  
Wait for next release.