LumaTeam/Luma3DS #497: Luma 8.0 region game patching hits black screen on previously unlaunchable game

SasinHyeonmu, 2017-06-05:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: (Clock+L2)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: (X )

Show NAND or user string in System Settings: (X )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

Launching Korean region version of Kirby: Planet Robobot (CTR-P-AT3K) on a non-Korean 3DS (ex. US) on Luma 8.0 leads to the user seeing a black screen.  The game will show the 3DS splash screen and look like it's launching, but will stick at a lit black screen with no sound and no option to use the Home button to return to the menu.  Only way to recover is to hold the power button until the 3DS shuts down.

This is different than Luma 7.1 behavior where Kirby: Planet Robobot would attempt launch and show 3DS launch splash screen, but then kick user to the menu and show an error (likely region lock error).

Issue will occur regardless of turning off 3DS clock and happens with game cartridge.

Korean region games that worked previously on Luma are unaffected.

**Steps to reproduce:**

1. Set up Boot9Strap 1.2 & Luma 8.0 on a non-Korean region 3DS
2. Insert Korean cartridge of Kirby: Planet Robobot (CTR-P-AT3K)
3. Set up region & language patching to Korean region and language
4. Reboot and attempt to launch the game cartridge.

**Dump file:**

Myriachan, 2017-06-05:
Try dumping title 0004009B00014202 from the NAND of a Korean 3DS and installing it on the non-Korean 3DS.  This is the Korean built-in system font.

Japanese, North American and European systems share a font; that's why this particular issue doesn't occur for them.

SasinHyeonmu, 2017-06-06:
I'll give that a try.  Have an Korean region old 3DS back on 8.x that I'll try and get bootstrapped.

CaiMiao, 2017-06-06:
I also mentioned maybe same problem in #541 ,
@TuxSH gave us a build that fixed **Language emulation**,
the classic `CHN ZH` or `KOR KO` should work on the build.
But the country/state emulation is still broken.

Akubon, 2017-06-07:
I have the same problem whit tomodachi life EUR in my american console

Hikari-chin, 2017-06-19:
@SasinHyeonmu Did @Myriachan's advice fix your issue?

closed 2017-06-22 by Hikari-chin