LumaTeam/Luma3DS #498: Luma 8.0 region game patching exceptions on launching previously unlaunchable game

SasinHyeonmu, 2017-06-05:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: (Clock +L2 / Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

Launching Korean region version of Tomodachi Life (CTR-P-EC6K) on a non-Korean 3DS (ex. US) on Luma 8.0 leads to the user hitting an exception.  The game will show the 3DS splash screen but then will hit the exception.

This is different than Luma 7.1 behavior where Tomodachi Life would attempt launch, show 3DS launch spalsh screen, but then kick user to the menu and show an error (likely region lock error).

Issue will occur regardless of turning off 3DS clock and happens with game cartridge.

Korean region games that worked previously on Luma are unaffected.

**Steps to reproduce:**

1. Set up Boot9Strap 1.2 & Luma 8.0 on a non-Korean region 3DS
2. Insert Korean cartridge of Tomodachi Life (CTR-P-EC6K)
3. Set up region & language patching to Korean region and language
4. Reboot and attempt to launch the game cartridge.

**Dump file:**
[crash_dump_00000003.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/1051140/crash_dump_00000003.dmp.zip)

HoLLy-HaCKeR, 2017-06-05:
Likely because of this in the [change log](https://github.com/AuroraWright/Luma3DS/releases)
>"Enable exception handlers" was removed, since it's always on with Rosalina choosing the best exception reporting method depending on the context

TuxSH, 2017-06-05:
Yep.

closed 2017-06-05 by TuxSH