LumaTeam/Luma3DS #499: N3ds xl back to luma7.1 after reboot

huygreat1996, 2017-06-05:
I can boot through luma 8.0 and set some setting, however after i reboot my n3dsxl the setting screen of luma7.1 appear. In other words, every time i need to use luma 8.0's new features i have to boot through it whenever i start my n3dsxl. How can i make it permanent?

Garro, 2017-06-05:
Isn't that because you have a Luma 8.0 on your SD while having Luma 7.1 on NAND?

If it's that the case this doesn't belong here as it is an user mistake, not Luma's.

huygreat1996, 2017-06-05:
Yeah, so how can i have luma 8.0 on NAND, my lumaupdater won't work

Margen67, 2017-06-05:
Copy boot.firm from the SD card to CTRNAND manually using GodMode9 or FBI.

BigBrainAFK, 2017-06-05:
https://3ds.guide/updating-b9s#section-iv---ctrnand-luma3ds

closed 2017-06-05 by TuxSH