LumaTeam/Luma3DS #500: [Minor Bug] Holding Rosalina Combination From Boot Freezes 3DS

shadowninja108, 2017-06-05:
**System model:**

New 3DS XL
11.4.0-37U

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9strap v1.2

Luma v8.0

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (3)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: (✔)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (✔)

Show GBA boot screen in patched AGB_FIRM: (✔)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

If I hold the Rosalina combination from boot up, it seems to never boot. The yellow communication never turns on. Forcing shutdown (holding power button for a while) then booting normally works perfectly fine. Not a severe bug, but a bug nonetheless.

**Steps to reproduce:**

1. Hold Rosalina combination (L + Down + Select)

2. Power it on

3. Watch the 3DS never boot

PatriqDesigns, 2017-06-05:
Can confirm

TuxSH, 2017-06-05:
The menu is probably not displaying given the screen aren't on. Press B to exit from the menu.

Gudf, 2017-06-05:
I just tried, and the 3ds still doesn't boot.

shadowninja108, 2017-06-05:
Pressed B periodically through boot (with Rosalina combination ), but it never advanced. That doesn't seem to be the problem.

Nanquitas, 2018-01-19:
Should be fixed with 7cb74b74d7c0e9dbc937b90d10600a1f07147c64 ;)