LumaTeam/Luma3DS #501: (REPRODUCABLE) An exception occurred - Data abort

PolarisBeaver, 2017-06-05:
**System model:**

[Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[11.4.0-37U SysNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap]

**Luma3DS version:**

[8.0]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: (x)

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
An exception occurred screen pops up when closing System Settings to return to the Home Menu after entering and exiting Nintendo DS Connections.

**Steps to reproduce:**

1. Launch System Settings

2. Press Internet Settings (and enter PIN if applicable)

3. Press Nintendo DS Connections, then OK to confirm launching

4. After the Nintendo DS Connections interface loads, press B or click the Back button to go back to System Settings

5. Press Back, and finally Close to close System Settings and return to the Home Menu.

6. Luma will pop up with the exception screen.

**Dump file:**

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1051237/crash_dump_00000000.zip)

HoLLy-HaCKeR, 2017-06-05:
Can reproduce on o3DS, 11.4.0-37E.
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1051750/crash_dump_00000000.zip)

sora10pls, 2017-06-05:
#453

TheMCNerd2017, 2017-06-05:
Was able to reproduce this glitch successfully on a N3DSXL running firmware 11.3.0-36U. 
[image]
Here is the dump file that was created.
[crash_dump_00000020.zip](https://github.com/AuroraWright/Luma3DS/files/1052119/crash_dump_00000020.zip)

sora10pls, 2017-06-05:
Reminder that my comment references an issue with this same problem, and that it happens with stock devices too...

shinji257, 2017-06-05:
I've also been able to reproduce this using the steps in the OP.  Personally, I would have never spotted this... I never go into that screen. >.<

Dump: [crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1052292/crash_dump_00000000.zip)

Screenshot:
[image]

Copy/pasteable version (yes I typed all of this)
```
An exception occurred

Processor:       ARM11 (core 0)
Exception type:  data abort
Current process: menu (0004003000008F02)

R0     00003FFF       R1     0032E950
R2     00000000       R3     00344FA0
R4     00000000       R5     00000002
R6     08000080       R7     00000001
R8     0032D5F8       R9     0032D680
R10    00000000       R11    00343650
R12    00065FF0       SP     0FFFFF58
LR     002330CC       PC     00117D9C
CPSR   20000010       FPEXC  00000005

You cna find a dump in the following file:
dumps/arm11/crash_dump_00000000.dmp

Press any button to shutdown

Stack dump:

0FFFFF58: 00 00 00 00 02 00 00 00
0FFFFF60: 80 00 00 08 CC 30 23 00
0FFFFF68: 00 00 00 00 03 D5 09 51
0FFFFF70: FF FF FF FF FF FF FF FF
0FFFFF78: 00 03 00 00 00 00 00 00
0FFFFF80: 01 00 00 00 38 38 02 00
0FFFFF88: 00 00 00 00 00 00 00 00
0FFFFF90: 00 00 00 00 00 00 00 00
0FFFFF98: 00 00 00 00 00 00 00 00
0FFFFFA0: 00 00 00 00 00 00 00 00
0FFFFFA8: 01 01 00 00 00 00 00 00
0FFFFFB0: F8 D5 32 00 78 02 34 00
0FFFFFB8: 01 01 00 00 01 00 00 00
0FFFFFC0: 00 00 00 00 00 00 00 00
0FFFFFC8: 00 00 00 00 54 50 10 00
0FFFFFD0: 00 00 00 00 00 00 00 00
0FFFFFD8: 00 00 00 00 00 00 00 00
0FFFFFE0: 00 00 00 00 3C 23 10 00
0FFFFFE8: 00 00 00 00 00 00 00 00
```

HoLLy-HaCKeR, 2017-06-05:
As @sora10pls mentioned, this is a known issue. See the issue he linked for an explanation. I think this issue can be closed.

sora10pls, 2017-06-05:
Is nobody gonna read my comment

TuxSH, 2017-06-05:
This is actually a bug in Nintendo's code, and want unoticed because HM was siletenly terminated.

Worked around (fixed) in latest commit anyways.

closed 2017-06-05 by TuxSH

Masterpokeagsg, 2017-06-06:
Currently updated to B9s 1.2 and luma 8.0 I appeared the error that many happened to them, like the one that creates a file called Crashdump in old 3ds Luma / dumps / arm11 / crash_dump_00000000.dmp
I can not take a photo I'm sorry :(

Margen67, 2017-06-06:
@Masterpokeagsg You need to use the latest nightly. https://astronautlevel2.github.io/Luma3DS/

Masterpokeagsg, 2017-06-07:
Thanks bro

jnozsc, 2017-06-17:
thanks everyone who worked on this issue.
It seems I met the similar issue when I try to setup luma by following this doc 
```
3ds.guide/installing-boot9strap-(browser)
```

I am not sure whether I should create a new issue or reply here.

here is the screenshot and dump file

[image]

[crash_dump_00000008.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/1082820/crash_dump_00000008.dmp.zip)

thanks!

jnozsc, 2017-06-17:
some addition info.

old 3ds ll, US region.
version 4.5 (If I don't misremember it)

jnozsc, 2017-06-17:
I have tried the latest public release version and nightly version, neither works

jnozsc, 2017-06-18:
sorry for previous post in this ticket, notice it is my side's issue. thanks

XiNaru, 2017-06-24:
That unfortunately did not help me .... I have the firmware 11.4.0-37E Luma 3DS 8 and boot9strap. Yesterday updated and came with me just in a game, after the use of an ntr cheats, and now the savegame is not right anymore.

Translated with google translator from the german

PolarisBeaver, 2017-06-25:
Ihre Spiel speichern Datei ist zerstört? Ruiniert? Das ist ein anderes problem Es ist nicht mit Luma3DS verwandt. Es ist ein Problem mit NTR.
Übersetzt mit Google

Your game save file has become destroyed? Ruined? That is a different problem. It is not related to Luma3DS. It is a problem with NTR.
Translated with Google

XiNaru, 2017-06-25:
But is the same error. Arm11: an exception occured. So I thought there was a connection between ntr and arm11, since I had before the update on boot9strap 1.2 and on luma 8 no problems with ntr had had.

Translated with google translator from the german

atom505, 2017-06-26:
can you help me also? 

I am have a problem with 

"you can find  a dump in the following file:
dumps/arm11/crash_dump_000000010.dmp

LiberoPat, 2017-07-09:
Has the issue pulled up by OP been fixed? I'm switching from Reinand and this same exact problem is occurring right now :/

Fuantei2, 2017-08-07:
> -ok I solved my issue, by pressing select and checking "auto-boot emunand" along with show in system settings, and now I'm good~

-update 2: I don't think I've successfully installed B9S due to this issue :/
-final: solved all issues by updating sysnand to latest firmware..

Same as you, except for the final solution, i cant update my SysNAND because i cant access it, when i try with no select "auto-boot emunand" it pop up me the error or if i try turning on the 3DS without SD card it shutdown inmediatly, i only can access the EmuNAND (thanks to you), what did you do to update your SysNAND¿, from the EmuNAND or you could access your SysNAND without problems¿

Sorry about my english

cactuar, 2017-08-17:
I have the same problem, I can only access my emunand. How did you update your sysnand? Thank you!

cactuar, 2017-10-12:
You can try to update your sysnad booting in recovery mode
https://3dbrew.org/wiki/Recovery_Mode
Use it at your own risk!

jeffdoho, 2018-02-06:
HI I have an issue with my new 3ds, it has a black icon, when I open it, it loads and says ( an exception occurred  arm11 (1) pre-fetch abort svc-break) any help please