LumaTeam/Luma3DS #503: InputRedirection problems

resetes12, 2017-06-05:
**System model:**

New3DS

**SysNAND version:**

11.4.0-37E SysNAND

**Entrypoint:**

Boot9Strap 1.2

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

**Explanation of the issue:**

- Enabling InputRedirection option and closing the 3DS lid crashes the console. 
- C-Stick and ZL/ZR buttons work on the menus, but not ingame. Tested on Monster Hunter XX and Smash Bros.
- ABXY are inverted on Xbox Controllers: A (3ds) is B (xbox), B is A, X is Y and Y is X.

Note that those two last issues may be caused by the InputRedirecion client, but I'm not sure so I'm reporting them just in case. The first one sure is.

**Steps to reproduce:**

1. Start InputRedirection with luma's menu.

2. Close the lid.

HoLLy-HaCKeR, 2017-06-05:
Cannot reproduce. Started Input Redirection, played around a bit (including MCU reboot), then tried closing and opening the lid. Everything still worked fine.

A is mapped to the 3DS' A button, so is B. (a bit counterintuitive, since they are at different places).

Tested on o3DS, on 11.4.0-37E, using TuxSH's client.

resetes12, 2017-06-05:
Maybe it's just a New3DS / New3DS XL thing.

Jan200101, 2017-06-05:
I have been able to reproduce it in another way affecting the normal Joystick

LycosHayes, 2017-06-05:
Same settings as OP, except SysNAND 11.4.0.37U.

I can confirm that zL/zR and C-Stick support is not working when in Monster Hunter Generations.

TuxSH, 2017-06-06:
I'll attempt fixing that around tomorrow. 😴

noscetarts, 2017-06-08:
I'm not yet updated on the cfw, but maybe, maybe in MH series games, activating the Control Pad option will effect. Maybe it behaves like a Circle Pad Pro, or maybe I'm wrong too.

LycosHayes, 2017-06-08:
by default my Circle Pad Pro is enabled in game, and is always enabled, tried turning it off then back on while having the InputRedirection on and still cannot use the C-Stick or zL/zR.

tekinlin, 2017-06-09:
Circle Pad Pro of old 3DS is not support. Hope it will be support.

Stary2001, 2017-06-11:
c-stick fixed (on n3ds) in 53e2ef2

closed 2017-06-16 by Hikari-chin