LumaTeam/Luma3DS #504: Rosalina not supporting "Title Select" Homebrews

TopKek69420, 2017-06-05:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-35E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( x)

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: ( x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

The newly introduced "Rosalina" doesn't support Homebrew apps that require you to select a title, like RomFSExplorer

**Steps to reproduce:**

1. Open Homebrew Launcher

2. Select any Homebrew that prompts a title select window

3. Error 

I know this seems to be intended/known, but I'd like to ask for support of this feature for convenience reasons. I know you can look at the RomFS/ExeFS of a game from GodMode9, but you need to know the TitleID for every game/have a list at hand. RomFSExplorer uses the Title select Window which shows the Game Name + Icon, making it way more convenient.

Clement95, 2017-06-05:
Up

fincs, 2017-06-05:
This is intentional. It was decided that target title selection is not useful anymore since 3dsx applications have full ARM11 system permissions now, and it is possible to directly access the savefiles of any title by using the appropriate FSUSER commands. In the meantime, you can use the CIA versions of savegame managers.

Also, you can use FBI to retrieve the title IDs of installed titles.

Clement95, 2017-06-05:
With this release, it works fine ! https://github.com/fincs/new-hbmenu/releases

closed 2017-06-05 by TuxSH